bigtreetech/PandaTouch #147: Feature request auto power off even when printer is not selected.

Diilandh, 2024-10-03:
Hello

I have 4 printers with 4 panda power bricks connected to a panda touch. The auto power off feature only works when I select the correct printer on the panda touch, then the timer runs and my printer shuts off.
For example i have the auto power off feature enabled on both my x1c and p1s and when they both are done printing only the printer that is selected on the panda touch shuts off automatically. The other one shuts off when i select the printer on the panda touch. When I select the group with all my printers then none off them shut down.

Thanks in advance 

Sincerely 

Dylan

dilixi, 2024-10-06:
> Hello
>
> I have 4 printers with 4 panda power bricks connected to a panda touch. The auto power off feature only works when I select the correct printer on the panda touch, then the timer runs and my [...]

Hi,
Could you please send me a video on this issue and tell me what kind of effect you want.

Email:
1284513393@qq.com
Thank you.

dilixi, 2024-10-09:
> BTW: same "problem" with the USB1 with the option to follow the chamber light. My P1S and the X1C are connected to one PandaTouch. If, for example, the P1S is currently selected on the PandaTouch, the PWR of the X1C does absolutely nothing. USB1 remains off when the printer's chamber lamp goes on until I select the X1C on the PandaTouch. I could live with that, but if I change the printer while printing on the PandaTouch, then USB1 on the PWR remains switched on when the printer is not selected. I have currently made a workaround for this via Node-Red and the PWR-API, but this is not optimal.
> 
> PandaTouch should monitor and evaluate all connected printers and PWRs in parallel. Otherwise, the multi-printer feature cannot actually be used in combination with PWRs.

Okay,  we will discuss this logic before the next release，thanks.

closed 2025-09-29 by xperiments