bigtreetech/PandaTouch #148: Panda Touch developped a restarting issue

philchfree, 2024-10-07:
Hi, I have one of the original panda touch without the new power board. I did not think I needed it as I have had few restarts and far apart. Panda Touch is now developping this issue more and more, do you think I need the upgraded power board (or tell me what component to add on the original board). Thx

bigtreetech, 2024-10-07:
> Hi, I have one of the original panda touch without the new power board. I did not think I needed it as I have had few restarts and far apart. Panda Touch is now developping this issue more and more, [...]

Will be easier to just replace the board. You can add a diode manually but it's just easier to ship the new board out to you. Please send a close up macro pic of your old board to luke@bigtree-tech.com as well as your original order num, shipping address and contact num.

closed 2024-10-07 by bigtreetech

philchfree, 2024-10-07:
Thank you, I will do that.