bigtreetech/PandaTouch #149: Request to add Panda Touch automatic restart function

redheadgiraffe, 2024-10-07:
If you connect the Panda Touch to the P1S and turn on the printer, the response speed is sometimes slow. (This may only apply to mine.) In that case, after RESTARTing Panda Touch, the response speed comes back quickly. So, it would be nice to have an automatic RESTART function once every 3 days or once every 5 days. I'd like to hear BIGTREETECH's thoughts on this.

bigtreetech, 2024-10-07:
Do you mean that it is slow to reconnect or that commands take a while to respond? We have greatly improved reconnect speed in 1.0.6 which should be in beta soon.

redheadgiraffe, 2024-10-08:
Commands are slow to respond. For example, when you touch to turn off the light, the Panda Touch displays the light off, but the printer light is still on. If you touch this repeatedly, the printer light turns off. In this case, if you restart Panda Touch, it will operate at its original speed again.

bigtreetech, 2024-10-08:
What firmware version are you using?

redheadgiraffe, 2024-10-08:
The latest version is version 1.0.5.

mytechguyri, 2024-11-13:
To piggy back on this.... I find my Panda Touch (v1.0.5.1 firmware) sometimes loses its WiFi connection, and the only way to get it to re-connect is to power cycle the Panda Touch.... I expect its because 2.4g networks trying to remain backwards compatible to 802.11b through 802.11ax and it just gets where it can't negotiate a connection, but a reboot clears everything and allows it to connect.  Would be nice if it had a watchdog so that if it sees a loss of WiFi connectivity for a certain period of time... 5 minutes perhaps?, it would automatically perform a restart....

bigtreetech, 2024-11-14:
> To piggy back on this.... I find my Panda Touch (v1.0.5.1 firmware) sometimes loses its WiFi connection, and the only way to get it to re-connect is to power cycle the Panda Touch.... I expect its [...]

The wifi connection actually already includes a full shut down and restart cycle. If it cannot find the printers for some time or if it cannot connect for some time it will shut the entire wifi hardware segment down and then kick start it. I suspect that your issue may even be somehow related to the router rejecting the connection attempt and needing the panda touch to back off for the time that the reboot takes.

gtgenter, 2024-11-14:
I had the same problems with losing wifi connectivity.  My problem was an outdated wife mesh system.  I updated my mesh system to an AX3000 and I have had no problems since upgrading the wifi mesh.   The outdated system was reaching the max number of connections on the mesh.  Old mesh was 36 connections and the upgraded mesh system can accommodate 160 connections.

redheadgiraffe, 2024-11-22:
So, do you have any plans to add an automatic restart function to Panda Touch? If the machine is turned on for a long time, unexpected errors may occur. Adding a restart function would be one of the ways to prevent a malfunction. I don't think it will be a difficult problem and I will wait for it to be applied in the next update.

xperiments, 2025-09-29:
Closing this issue as it has been resolved in the latest versions

closed 2025-09-29 by xperiments