bigtreetech/PandaTouch #150: Meta-request: please use the Github 'Releases' functionality for better notifiabiliy

a10kiloham, 2024-10-20:
Sites like https://newreleases.io/ track when there is a new release added and can notify. It's hard to tell when you add a new release here otherwise, it requires setting a crawler up on your wiki.
Thank you!

BTTJUSTIN, 2024-10-23:
Thanks for making this suggestion. We will discuss it with our engineer.

UsedOak, 2024-10-24:
Yesssss

dilixi, 2024-12-24:
> Sites like https://newreleases.io/ track when there is a new release added and can notify. It's hard to tell when you add a new release here otherwise, it requires setting a crawler up on your wiki. Thank you!

Hi,
Did you have time to do a test for this?
we can do a test with this rep:
https://github.com/bigtreetech-beta/PandaTouch-Support

And please subscribe to events for bigtreetech-beta/PandaTouch-Support

[image]

You can email me when you have time.

1284513393@qq.com

a10kiloham, 2024-12-24:
Yes I can do that notification but I only want notification when there is a release not idle chatter. You still haven't set up releases yet properly.

dilixi, 2024-12-25:
> Yes I can do that notification but I only want notification when there is a release not idle chatter. You still haven't set up releases yet properly.

I mean that you need to do something before you can get this notification.
you need to follow these steps:
1 open this link
https://github.com/bigtreetech-beta/PandaTouch-Support

2 click the 'watch' button and click the 'custom' button
[image]

3 choose the 'releases'
[image]

4 give me a email and I will add a release on PandaTouch-Support

5 check your email

I think it is very easy for you to do that test.

dilixi, 2024-12-25:
> Yes I can do that notification but I only want notification when there is a release not idle chatter. You still haven't set up releases yet properly.

@a10kiloham 

I mean that you need to do something before you can get this notification.
you need to follow these steps:
1 open this link
https://github.com/bigtreetech-beta/PandaTouch-Support

2 click the 'watch' button and click the 'custom' button
[image]

3 choose the 'releases'
[image]

4 give me a email and I will add a release on PandaTouch-Support

5 check your email

I think it is very easy for you to do that test.

maple-underscore, 2025-02-13:
Any update on this? I also think that releases would be a better system than having to go through the firmware folder to see if there is a new one.

bigtreetech, 2025-02-13:
Should be coming with 1.0.7 @dilixi