bigtreetech/PandaTouch #153: Bambu-Cloud Login not working if 2FA is enabled

BOFH90, 2024-10-26:
I cannot login to my BambuLab Account if 2FA is enabled for the account.
Error message: _Login failed, please check network._

I´m on the latest firmware (1.0.5).

burlandp, 2024-11-06:
I also have same issue on my new Panda, I tried 1.0.4 and 1.0.5 neither worked.
I was trying to login to "Login BambuLab Account" using GLOBAL region. My BambuLab Account is actually attached to ASIAPACIFIC region, but his region does not appear under Panda Touch

dilixi, 2024-11-19:
> I cannot login to my BambuLab Account if 2FA is enabled for the account. Error message: _Login failed, please check network._
> 
> I´m on the latest firmware (1.0.5).

@BOFH90 
Hi,
Can you login correctly with V1.0.5.1? Because the BBL change the method of login.

if everything is alright please close this issue , thanks.

BOFH90, 2024-11-19:
I´m still on the V1.0.6 Beta. Is 2FA-Login also possible with the 1.0.6 release?

dilixi, 2024-11-19:
> I´m still on the V1.0.6 Beta. Is 2FA-Login also possible with the 1.0.6 release?

If you enable the 2FA, we need to do it another new version.

nmfireman, 2024-11-26:
I just received my Panda Touch and can't log into my Bambu Labs account. This pretty makes this device unusable. Has there been a fix for this I do have 2FA turned off

spartanrob, 2024-11-26:
Same issue here!

nmfireman, 2024-11-26:
Well at least I bought mine from Amazon so back it will.

spartanrob, 2024-11-26:
Update: just updated the panda touch to the latest firmware and am able to log in to Bambu

nmfireman, 2024-11-26:
What firmware and where can I get it

Greg Bouten

dilixi, 2024-11-26:
> What firmware and where can I get it Greg Bouten
> […](#)

@nmfireman 
Hi,
please have a try with this one:
https://github.com/bigtreetech-beta/PandaTouch-Support/tree/main/V1.0.6

nmfireman, 2024-11-26:
Sorry pretty new at this could you please leave a link to that firmware

nmfireman, 2024-11-26:
So I download the bin file and go to the Panda IP address and update there?

spartanrob, 2024-11-26:
Yes. I needed the bin and the img file. Was a 2 step process

nmfireman, 2024-11-26:
So first bin and then image file? I’m looking for how to update firmware now

Greg Bouten

dilixi, 2024-11-26:
@nmfireman 

https://bttwiki.com/PandaTouch.html#how-to-update-firmware

nmfireman, 2024-11-26:
Thank you. You guys are great.

Greg Bouten

dilixi, 2024-11-26:
> Thank you. You guys are great. Greg Bouten
> […](#)

@nmfireman 
Does it works well now?

nmfireman, 2024-11-26:
No sorry it’s late for us. I will work on this tomorrow. I just appreciate
people showing me the way on some of these things as I’m not very up to
speed on it.

Greg Bouten

BOFH90, 2024-11-26:
I can also confirm the Login works with 2FA enabled on the new firmware 👍

closed 2024-11-26 by BOFH90

nmfireman, 2024-11-26:
Ok, I could connect with FA2 turned off, but it seems to be having issues
when I turn it on. I had it setup with the google authenticator app on my
Iphone. I'm good with not using FA2 as it has been that way since day one I
did have some issues with the P1P usb port. I was working fine but after
the update I seemed like it wanted more power as the screen was flashing on
and off. I put an adapter on the Type A end of the provide USB cable and
plugged it in to a wall brick and it worked fine. then when ti tried it
back in to the p1p it also worked just fine. Not sure what that was about.
Thanks for all the pitched in to help. I hope
 I can help you some day.
Greg Bouten ***@***.***