bigtreetech/PandaTouch #154: No back option in certain menues.

mytechguyri, 2024-10-28:
Please add a "back" option on certain multi-depth menus.   For example, when one goes into the "temperature / axis" menu, (layer 1) and then drills down to the fan speed control menu (layer 2).   There's no way to back out of that 2nd layer menu and return to layer 1 to  adjust the axis settings other than power cycling the screen.   There should be a "back" button/icon on the 2nd layer menu to allow the user to back out to the 1st layer menu.
[image]
[image]

dilixi, 2024-10-30:
> Please add a "back" option on certain multi-depth menus. For example, when one goes into the "temperature / axis" menu, (layer 1) and then drills down to the fan speed control menu (layer 2). [...]

Hi, you can press the left panel to back to the parent page.

[image]

mytechguyri, 2024-11-01:
but that doesn't bring you back to the parent page, pressing the left panel only brings you back to the 2nd level menu you were already in.

mytechguyri, 2024-11-01:
I did discover pressing the N/A C button brings you back to the previous level menu for some reason... not very intuitive.

dilixi, 2024-11-04:
> I did discover pressing the N/A C button brings you back to the previous level menu for some reason... not very intuitive.

Hi:
1 press the left button to the speed control panel
2 press the left button again to return to parent panel

https://github.com/user-attachments/assets/91e8d5d8-314c-4a89-8d5c-4e5e4b20f8f1

mytechguyri, 2024-11-05:
Ok... Not very intuitive, and could be improved, but that works

slothy89, 2024-11-10:
You literally tap on the Fax control section again, and it goes back to the layer below. can't get more intuitive than that.

https://github.com/user-attachments/assets/4d67cd30-1934-431a-b3cd-d7f6e9cfb338

bigtreetech, 2024-11-11:
Closing for now as the UI is fairly intuitive as it currently stands. Adding extra buttons too the UI to go back would clutter it and likely be more confusing. The colour of the menu option is highlighted when the UI elements for that menu option are displayed. Tapping it again removes the highlight and goes back to the default.

closed 2024-11-11 by bigtreetech