bigtreetech/PandaTouch #155: Feature Request: add filament drying menu option like the Bambu X1C has.

mytechguyri, 2024-11-05:
While the Bambu Wiki says the P1S can dry filament too, just like the X1C, it is a manual process.  Adding the menu in the  Panda Touch to give X1C like functionality would be a nice feature.
https://wiki.bambulab.com/en/general/bambu-filament-drying-cover#:~:text=If%20you%20need%20to%20dry,dry%20them%20for%2012%20hours.

bigtreetech, 2024-11-05:
I like this! @dilixi This should be easy to implement. Let's make it a priority for 1.0.7.

Eevilmouse, 2024-12-06:
> While the Bambu Wiki says the P1S can dry filament too, just like the X1C, it is a manual process. Adding the menu in the Panda Touch to give X1C like functionality would be a nice feature. https://wiki.bambulab.com/en/general/bambu-filament-drying-cover

This would make me buy the BTT Panda touch. As for right now, I cant justify the cost just for AMS control (which I have on the app)