bigtreetech/PandaTouch #156: Panda Pwr - Power outage issue

Tinkerguy33, 2024-11-05:
If you lose power in my case to my house. When Power comes back on the Panda Pwr Will power the printer back up. This can go unnoticed if setting in another part of the house. So it defeats the purpose of using it.

bigtreetech, 2024-11-05:
This was an intentional design decision. We are unable to tell whether power has been out for a minute or for months. On devices that were turned off and then plugged back in perhaps months later, users expect there to be power. If they plug it in and forget that the last state was "power off" then often it is assumed that the product is faulty because there is no power to the machine through it.

We are open to considering memory retention through power cycles though.

@dilixi lets discuss this tomorrow.

Tinkerguy33, 2024-11-06:
Hmmm. They should do the same concept with all appliances.

Tinkerguy33, 2024-11-06:
I guess that is why computer bios lets you decide power state after power loss.  For a casual user with the printer in spare bedroom, I don't want to be going in that room everyday to see if the printer is on. I guess I will have to remove it and go back to using my home automation to detect power consumption and have it turn it off. Not everyone has a home automation system that can do that.