bigtreetech/PandaTouch #157: X1C with 3 AMS

Buyco-dotcom, 2024-11-06:
A question for the Panda Touch users: I recently bought a third AMS and found that as soon as 3 AMSs are connected to the X1, the MQTT connection does not work properly. The funny thing is that everything can be controlled, but it doesn't show anything regarding status, temperature, etc. When I unplug an AMS everything works fine. My BLLED controller doesn't cause any problems, but neither does the connection to the home server. It may just be because the power supply is not sufficient.

Buyco-dotcom, 2024-11-06:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Buyco-dotcom, 2025-03-23:
With the new firmware update for Bambulab X1C 01.08.05.00, the Panda Touch will no longer be usable if more than two AMS units are connected to the printer. Since data is only transferred in cloud mode and not in LAN mode, enabling developer mode on the printer is not an option.

bigtreetech, 2025-03-23:
Hey. Thanks so much for taking the time to report. We are already aware of the issue and a fix will be pushed this week. Simple one to solve.

bigtreetech, 2025-03-23:
OK great. I thought that it may have been solved already in 1.0.7.

closed 2025-03-23 by bigtreetech