bigtreetech/PandaTouch #159: Login Failure

rashuge, 2024-11-08:
I do not have 2FA configured for the Bambu account.  I have followed the instructions online that indicate that I have to set the password for my account, I have followed that instruction set.  I have confirmed with a private browser that I can login with the password instead of the Google account.  I have to still do an email verification, however when logging in via a private browser session.  I cannot get it to authenticate.  I have global set as my region.

Panda Touch Version: 1.0.5

dilixi, 2024-11-08:
> I do not have 2FA configured for the Bambu account. I have followed the instructions online that indicate that I have to set the password for my account, I have followed that instruction set. I have [...]

Hi,
Please use this firmware:
https://github.com/bigtreetech-beta/PandaTouch-Support/tree/main/V1.0.6

rashuge, 2024-11-08:
That worked amazingly.  Perfect.  Thank you so much.

BTTJUSTIN, 2024-11-20:
This bug was fixed in FW version V1.0.6

closed 2024-11-20 by BTTJUSTIN

pithon55, 2025-02-07:
404 - page not found

pithon55, 2025-02-07:
Update link

https://github.com/bigtreetech-beta/PandaTouch-Support/tree/main/1.0.6.3