bigtreetech/PandaTouch #166: Camera feed

galileoleosolis, 2024-11-09:
I know it has been asked and it may not be possible, but please please please add some way to see what's going on, even if it grabs a still image and puts it on the screen.

maple-underscore, 2025-01-31:
Would there be some issues with the new firmware update?

maple-underscore, 2025-02-12:
Yeah, especialy since the private keys to the auth system have already been leaked.

maple-underscore, 2025-02-12:
> The private keys are less critical since we expect that certificate to be blacklisted immediately. The 32 byte app secret was really the highlight of that situation since that offers the key to request any certificate at all. We also expect that to be changed and for the security around it to be hardened significantly. Nevertheless, it is very rare that secrets distributed in software remain secrets unless they are contained within some sort of hardware based secure enclave. It will be interesting to see what the community comes up with after the next update to the firmware/apps.

Wasn't there something about the printer certificate expiring after 1 year of use? Or is that not true? Also, IMO it's pretty obvious that Bambu is just trying to restrict the closed-source environment even more, but that's just my opinion.

bigtreetech, 2025-02-13:
To avoid retyping a lengthy account that details the facts I will paste two blog post articles which detail what went down:

https://bigtree-tech.com/blogs/news/pandas-love-bamboo
https://bigtree-tech.com/blogs/news/announcement-on-bambu-lab-security-firmware-update