bigtreetech/PandaTouch #175: i2c connector : temperature not updating

jlo1975, 2024-11-20:
hello
getting the brand new fresh firmware to finally use the long awaited i2c connector support.
But sadly,  the temperature is not updating (you have to reboot the device to get only 1 new reading)
for info the humidity is updating correctly.

jlo1975, 2024-11-22:
some update.
after few reboot the temperature reading update correctly every seconds.
for info here the sensor i'm using
AS-delivery dht-20 i2c (temp and humidity sensor)

looxonline, 2024-11-24:
The port is designed to work with the new Panda Sense module that will be released shortly. We cannot guarantee operation with other modules.

closed 2024-11-24 by looxonline

TeoMM, 2024-11-29:
@jlo1975 could you take a picture to show us how you connected it? ofc it is not recommended by the manufacturer but im curious just how you made it work and how you wired it. Thanks!

jlo1975, 2024-11-29:
i've just done a 1 to 1 wiring with the info behind the pandatouch here -> https://bttwiki.com/img/PandaTouch/pt_back.png and the info on the sensor

jlo1975, 2024-11-29:
i was lucky to found in my hardware box mess the good little white connector on pandatouch side