bigtreetech/PandaTouch #180: Increase the size of the Thumbnail Picture

macman-3D, 2024-11-23:
Just installed Firmware v1.0.6 on my Panda Touch, so far works flawless for my needs. I have a suggestion/enhancement, can the thumbnail picture of the print be  made larger on screen in future release?

BTTJUSTIN, 2024-12-05:
Thanks for making this suggestion. We will discuss it internally and let you know if it will be scheduled for a future firmware release.

heelsmj, 2024-12-07:
I second this one. Thank you so much for the new thumbnail feature it is amazing! If you could make the image bigger that would be incredible thanks! It is so cool to have a product that is being improved upon regularly!

dilixi, 2025-04-11:
Hi all, 
Please have a try with this one.
The printing task need to be started by bambu studio or handy app. 

https://github.com/bigtreetech-beta/PandaTouch-Support/blob/main/firmware/v1.0.7.1_beta4_big_thumbnail.bin

macman-3D, 2025-04-11:
> Hi all, Please have a try with this one. The printing task need to be started by bambu studio or handy app.
> 
> https://github.com/bigtreetech-beta/PandaTouch-Support/blob/main/firmware/v1.0.7.1_beta4_big_thumbnail.bin

Much Better on the thumbnail - Any possible way it can a bit larger or is this the max level? Thanks again!

[image]

dilixi, 2025-04-14:
> Much Better on the thumbnail - Any possible way it can a bit larger or is this the max level? Thanks again!
>
> [image] [ [image] [...]

@macman-3D 
Hi,
The max size is 280*360.
please have a try with this one:
https://github.com/bigtreetech-beta/PandaTouch-Support/blob/main/firmware/v1.0.7.1_beta5.bin

macman-3D, 2025-04-14:
Much Much Better Jonny - thanks so much for doing this.

[image]

closed 2025-04-14 by macman-3D