bigtreetech/PandaTouch #181: Temperature control still not working?

cjfury, 2024-11-23:
Hi there, I had FW 1.0.5 and just upgraded to 1.0.6. I had issues where I could not control the temperature on the screen previously but seems to still be persistent. Everything else worked perfectly with zero issues, I could move nozzle, bed, speed, fans - just no bed or nozzle temp changes. I could punch in the desired value and it would sit there with an orange icon as if it is heating up but no dice. I flip the screen up to view what the original P1S screen is showing and it's got change, sits at 38/0 rather than 38/200 and heat up.

Any advice or others have similar issues?

dilixi, 2024-11-24:
> Hi there, I had FW 1.0.5 and just upgraded to 1.0.6. I had issues where I could not control the temperature on the screen previously but seems to still be persistent. Everything else worked [...]

Hi,
You need to login a account,BBL did change something since V1.0.4

looxonline, 2024-11-26:
> Hi there, I had FW 1.0.5 and just upgraded to 1.0.6. I had issues where I could not control the temperature on the screen previously but seems to still be persistent. Everything else worked [...]

This is covered in the wiki. In particular, check the important points section right at the start of the wiki article on the panda touch and then later the cloud mode section which explains how to get set up with cloud mode. You must log into cloud mode in order to control heaters.

closed 2024-11-26 by looxonline