bigtreetech/PandaTouch #191: Filament control: load and unload do the same - unload

Nobiscum, 2024-12-08:
Printer:     Bambu P1S. 
Firmware: 1.0.6.

Problem: When I try to unload and load a new filament (external spool) Panda sends the same command to the printer - unload filament. Both buttons (load and unloads) do the same - unloading. To load new filament I have to use native control on printer or from bambu studio.

Debigarey, 2024-12-16:
Same thing is happening to me.

dilixi, 2024-12-17:
@Nobiscum  @Debigarey 

I did a unload and a load test , it works well, 
How to reproduce this problem? 

What is the firmware version of your printer and Panda Touch?

Debigarey, 2024-12-17:
Hello, I was actually about to post another comment to mention the fact that upgrading the p1s firmware  to 1.07 fixed the issue.It is working correctly now. Thank you for the quick reaction.Yves

Nobiscum, 2024-12-17:
@dilixi unfortunately I still have the same problem even after upgrade to 1.07. bambu firmware. How to reproduce:
Load PETG-HF. 
Unload.
Change filament type to TPU-HF
Try to Load. 

I also reset the configuration to default and configured PandaTouch again. No luck. I'm going to upload and share a demo a bit later.

dilixi, 2024-12-18:
> @dilixi unfortunately I still have the same problem even after upgrade to 1.07. bambu firmware. How to reproduce: Load PETG-HF. Unload. Change filament type to TPU-HF Try to Load.
> 
> I caught that: So the problem comes when you switch fillament type between PETG-HF and TPU-HF and vise versa (in my case at least). Load PETG-HF, wait a bit, unload PETG-HF, change filament type to TPU-HF, load TPU-HF, unload TPU-HF, change filament type to PETG-HF, load.
> 
> I also reset the configuration to default and configured PandaTouch again. No luck.

Hi, 
If you reboot the printer when you change the filament type , dose it can works well?

dilixi, 2024-12-18:
> @dilixi unfortunately I still have the same problem even after upgrade to 1.07. bambu firmware. How to reproduce: Load PETG-HF. Unload. Change filament type to TPU-HF Try to Load.
>
> I caught that: So [...]

@Nobiscum 
Hi,
If you reboot the printer when you change the filament type , dose it can works well?

dilixi, 2024-12-18:
> @dilixi unfortunately I still have the same problem even after upgrade to 1.07. bambu firmware. How to reproduce: Load PETG-HF. Unload. Change filament type to TPU-HF Try to Load.
>
> I caught that: So [...]

Hi,
I find the real reason,thanks.

Debigarey, 2024-12-18:
> Hi, I find the real reason,thanks.

Hello, can you share the solution please. Thank you.

dilixi, 2024-12-19:
> Hello, can you share the solution please. Thank you.

Yes,
Please send me a email, I need to confirm that version is ok.
1284513393@qq.com

Nobiscum, 2024-12-20:
@dilixi panda_touch-v1.0.6.2a.bin on this firmware I was not able to reproduce bug. Seems it's fixed. Thanks.

Nobiscum, 2024-12-20:
> Oh, there's a new firmware? @Nobiscum

Just a test build to test the issue with loading fillament.

dilixi, 2024-12-20:
> I have the same issue when loading different colors of PLA, these are the steps i follow
> 
> 1. Unload
> 2. Load (fails)
> 3. Load (fails)
> 4. Load (works)
>    really annoying to have to load 3 times, I'll try the test build this afternoon!

Thanks,
This issues has been fixed with 1.0.6.2.

closed 2024-12-21 by dilixi