bigtreetech/PandaTouch #192: Issue launching a print using AMS from PandaTouch

Benigans, 2024-12-08:
Hi,

I use OrcaSlicer to prepare my prints. I usually click "Print" and it's working great. On the panda touch I see the print progresse and everything.

But, when I want to print it again using the file browser, I see it on the list (and the thumbnail) but the print is stuck just before loading the correct filament from the AMS (heatbed preheating).
It stays there indefinitely and never load any filament.

When I launch the same print from the builtin screen (the basic one from the P1S) the print works fine.

It's the same when I "send" from OrcaSlicer a multicolor print or a single color, as long as it uses the AMS.
On the Panda, the AMS is recognized, the correct filaments are at the correct places.
I read the docs and it doesn't seem to be a limitation (like the multiplate integration).

Is this a normal behavior ?
Thx

Benigans, 2024-12-17:
Anyone having the same issue ?

dilixi, 2024-12-17:
> Anyone having the same issue ?

Hi,
What is the firmware version of your printer and your panda touch?

Benigans, 2024-12-18:
Printer is 1.07.00.00 and Panda is 1.0.6.1
I updated the printer and the ams yesterday to see if this was the cause but no change.

Benigans, 2024-12-18:
One thing to note is that if in the 3mf file I selected the first spool on the AMS, the print starts.
As long as I referred to A2/3/4 on the 3MF, the print doesn't work from the Panda.
I also tried to select A3 next to the "use AMS" checkbox but that doesn't work.

And from the builtin screen I juste launch the print without doing anything else and it works

Benigans, 2024-12-20:
I tried the recovery tool on the Panda but that didn't change.
I updated to 1.0.6.2 : no change.

Just to be sure, I tried to print the same files from the web app "BambuFarm" : https://github.com/TFyre/bambu-farm, and it worked.

dilixi, 2024-12-21:
> One thing to note is that if in the 3mf file I selected the first spool on the AMS, the print starts. As long as I referred to A2/3/4 on the 3MF, the print doesn't work from the Panda. I also tried [...]

Hi,
I did select A3 and the printer works well.
Can you send me a email with detail steps of reproducing it 100%.(1284513393@qq.com)
My steps:
1 go to the sdcard panel 
2 start a printing task by use ams and select A3 
3 printer works well with AMS

Vantage-DS, 2024-12-22:
> Hi,
>
> I use OrcaSlicer to prepare my prints. I usually click "Print" and it's working great. On the panda touch I see the print progresse and everything.
>
> But, when I want to print it again using the [...]

Just started having the same issue right now. It's been working fine and now without changing anything it's just stopped.

Goes to print, won't load filament... just sits there warm and ready, but nothing.... no error message, not showing as paused.

Initially it just did it on 1 of my printers. I actually did a factory reset on it because I thought the AMS had corrupt itself.

Printing from the SD card works absolutely fine.
Printing from the Panda Touch does not.

Tried updating Panda firmware too.

P1S firmware: Both 1.07.00.00 and 1.06.01.00 have the issue.
AMS Firmware: most recent and 00.00.06.44

dilixi, 2024-12-22:
> Just started having the same issue right now. It's been working fine and now without changing anything it's just stopped.
>
> Goes to print, won't load filament... just sits there warm and ready, but [...]

Thanks,I will check it out tomorrow.

dilixi, 2024-12-23:
> Just started having the same issue right now. It's been working fine and now without changing anything it's just stopped.
>
> Goes to print, won't load filament... just sits there warm and ready, but [...]

Hi,
Please have a try with auto.
which region did you select when you bind a printer?
https://github.com/bigtreetech-beta/PandaTouch-Support/blob/main/1.0.6.2/panda_touch-v1.0.6.2c.bin

TechTinkerLab, 2025-01-10:
I think that i have very similar
 bug. I'm using a Bambu Lab P1S with AMS. Today I got my Panda Touch. I updated firmware on Panda Touch to the newest version - v1.0.6.3. Then I added my P1S in Panda Touch as a LAN printer. Everything seems to be ok. From Panda Touch, I can see print files on the SD card, I can load and unload filaments in AMS, turn on and turn off LEDs. But when I try to print a file via Panda Touch that I sent from Orca Slicer, the print doesn't start. It shows me "--.3mf" as filename, 0 time, 0%. Nothing. It just stays in this state. I also checked it with other files. The screen is not frozen. I can still turn on and off lights, check SD card content. In Orca Slicer and in default LCD, I can't see that something is printing. Panda Touch stayed in this state for at least an hour and nothing changed. A I also tried exporting the file to a USB stick and then selecting it to print in Panda Touch from the USB stick. Unfortunately, the result is the same. When I try to print from Orca/Bambu Studio, I can see on Panda Touch that my file is being printed - temperatures etc. are updating. I can even pause/stop the print from Panda Touch. Only starting a print from it fails for some reason. In attachment: screen from stock lcd and Panda touch LCD after starting printing from Panda Touch. 
Panda touch firmware -  v1.0.6.3.
Bambu LAB P1S firmware - 01.07.00.00
AMS firmware - 00.00.06.49
[image]
[image]

dilixi, 2025-01-10:
> I think that i have very similar bug. I'm using a Bambu Lab P1S with AMS. Today I got my Panda Touch. I updated firmware on Panda Touch to the newest version - v1.0.6.3. Then I added my P1S in Panda Touch as a LAN printer. Everything seems to be ok. From Panda Touch, I can see print files on the SD card, I can load and unload filaments in AMS, turn on and turn off LEDs. But when I try to print a file via Panda Touch that I sent from Orca Slicer, the print doesn't start. It shows me "--.3mf" as filename(it should be 3dbenchy), 0 time, 0%. Nothing. It just stays in this state. I also checked it with other files. The screen is not frozen. I can still turn on and off lights, check SD card content. In Orca Slicer and in default LCD, I can't see that something is printing. Panda Touch stayed in this state for at least an hour and nothing changed. A I also tried exporting the file to a USB stick and then selecting it to print in Panda Touch from the USB stick. Unfortunately, the result is the same. When I try to print from Orca/Bambu Studio, I can see on Panda Touch that my file is being printed - temperatures etc. are updating. I can even pause/stop the print from Panda Touch. Only starting a print from it fails for some reason. In attachment: screen from stock lcd and Panda touch LCD after starting printing from Panda Touch. Panda touch firmware - v1.0.6.3. Bambu LAB P1S firmware - 01.07.00.00 AMS firmware - 00.00.06.49 ![panda_touch_lcd_after_starting_print_from_it](https://private-user-images.githubusercontent.com/170227641/402088148-fd2df7e6-d036-47e6-b152-33678013b31f.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY1NDYxOTAsIm5iZiI6MTczNjU0NTg5MCwicGF0aCI6Ii8xNzAyMjc2NDEvNDAyMDg4MTQ4LWZkMmRmN2U2LWQwMzYtNDdlNi1iMTUyLTMzNjc4MDEzYjMxZi5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMTEwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDExMFQyMTUxMzBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wN2Y5YWMzNTljYWMyZWQ5M2NmZjg4OWU4ZmJjMTg1ZTY0N2EwN2EzOTIyNWU4NWM3NWJkNjdkZmNmNGI2ZjZjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.po85--Gw0eDNGxEqSdysLDbBhWKALZdpKonUoI8-hp0) ![stock_lcd_after_staring_print_from_panda_touch](https://private-user-images.githubusercontent.com/170227641/402088158-0938db9e-5084-45d5-b177-44b4db006129.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY1NDYxOTAsIm5iZiI6MTczNjU0NTg5MCwicGF0aCI6Ii8xNzAyMjc2NDEvNDAyMDg4MTU4LTA5MzhkYjllLTUwODQtNDVkNS1iMTc3LTQ0YjRkYjAwNjEyOS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMTEwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDExMFQyMTUxMzBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lYjk0YmNiMGJhOWVhMmI1OTdlYzE3MzYxNGYwMjNiMjVhMjg5MzYxYTgwMTc2ZDFjMjI5MmYxMjJjZTU2NDEzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.7U1BYr-RzbqYDT0Bh-tl-AfDnoqaQ7_7aCLLEJwKOik)

Hi,
Please login an account when you are going to control the printer to start a printing task.

TechTinkerLab, 2025-01-11:
So, I followed your advice and it works. Thank you! It turns out it was my mistake, not a bug in the software. :)

Benigans, 2025-02-28:
Hi,

you sure it's fixed ?
I'm still on a beta firmware 1.0.6.2 "nomap" to mitigate the issue [This one](https://github.com/bigtreetech-beta/PandaTouch-Support/tree/main/1.0.6.2)
TechTinkerLab issue was a different issue.
1.0.6.3 changelog doesn't seem to resolve my issue.

Thx