bigtreetech/PandaTouch #194: New firmware version 01.07.00.00 for P1S released

macman-3D, 2024-12-10:
Bambu released an update firmware and wondering if it's compatible with Panda Touch Firmware Version 1.0.6? Attached are the release notes from Bambu - thanks!

version 01.07.00.00 update information:
# Version 01.07.00.00
## New Feature
1. Added support for offline Firmware update from MicroSD card.
2. Added support for filament editing while printing.
## Feature Optimization
1. Pause print if abnormal bed-leveling data exceed the threshold. This helps protect the buildplate from unwanted damage.
2. Optimized Ul Copy.
## Bug Fixed
1. Fixed the printer screen freeze problem when user using Bambu Handy to initialize the first calibration
2. Fixed the printer restarting problem caused by saving the K-factor values.

macman-3D, 2024-12-11:
All works good! Updated to Panda Touch v 1.0.6.1 with no issues to report from what I'm experiencing.

looxonline, 2024-12-11:
Works all good. Feel free to upgrade.

closed 2024-12-11 by looxonline