bigtreetech/PandaTouch #199: Panda Touch Stuck in "OTA Not finished yet" After attempt to flash with Firmware Tool

wsturner88, 2024-12-15:
My Panda Touch is stuck in "OTA is not finished yet".   When I select a wifi it reboots every time.

Tried Connecting with the Recovery_toolV1.0.6 2 but it has lost the ability to talk to the Panda Touch.

I was able to connect to it once, before I had this issue, now no luck.

Any Help would be greatly appreciated.

marioschaniel, 2024-12-15:
Had a similar issue. The URL was <ip-address-of-panda-touch>:8080 with a file upload form. That one is the wrong one (accepts only the img file). 

I removed the 8080 from the url and got the main screen. From there I uploaded the firmware (bin) through the Update FW button and afterwards the img file through the Update File button - then it worked again.

dilixi, 2024-12-15:
> My Panda Touch is stuck in "OTA is not finished yet". When I select a wifi it reboots every time.
>
> Tried Connecting with the Recovery_toolV1.0.6 2 but it has lost the ability to talk to the Panda [...]
Hi,
@wsturner88

1 the panda touch doesn't connect to the ap?

2 can open a ap which use a same ssid from your phone?

3 can you get me more description about "lose the ability to talk to panda touch"

4 Please send me a email via 1284513393@qq.com

> My Panda Touch is stuck in "OTA is not finished yet". When I select a wifi it reboots every time.
>
> Tried Connecting with the Recovery_toolV1.0.6 2 but it has lost the ability to talk to the Panda [...]

> My Panda Touch is stuck in "OTA is not finished yet". When I select a wifi it reboots every time.
>
> Tried Connecting with the Recovery_toolV1.0.6 2 but it has lost the ability to talk to the Panda [...]
Hi,
@wsturner88

1 the panda touch doesn't connect to the ap?

2 can open a ap which use a same ssid from your phone?

3 can you get me more description about "lose the ability to talk to panda touch"

4 Please send me a email via 1284513393@qq.com.

wsturner88, 2024-12-15:
Panda Touch won't connect to WIFI so I can't access it that way.   It reboots the second I select the SSID.   The Recovery app no longer connects.   Is there a way to put it in DFU or BOOT MODE?

dilixi, 2024-12-15:
> Panda Touch won't connect to WIFI so I can't access it that way. It reboots the second I select the SSID. The Recovery app no longer connects. Is there a way to put it in DFU or BOOT MODE?

"The Recovery app no longer connects."

I don't understand.

You don't send me any message from email.

wsturner88, 2024-12-15:
SENT.. Thanks for the Help..

dilixi, 2024-12-15:
> SENT.. Thanks for the Help..

I don't know the type of your computer, but you can check out with these video first.

https://github.com/bigtreetech-beta/PandaTouch-Support/blob/main/use_recovery_tool_with_typec_charger_cable.mp4

https://github.com/bigtreetech-beta/PandaTouch-Support/blob/main/recovery_tool_for_macos.mp4

wsturner88, 2024-12-15:
The MAC instruction worked..  I have the Panda updated..  Couple of changes.
MAC Sequoia - After you install the ESP Driver go to Settings - Login Items & Extensions - Driver Extensions and enable the new CH34x Driver - REBOOT
[image]

Thorny Instructions:
Open Thorny - go to the RUN Menu
Select Configure Interpreter 
[image]

Follow the Rest of the instructions

dilixi, 2024-12-15:
> The MAC instruction worked.. I have the Panda updated.. Couple of changes. MAC Sequoia - After you install the ESP Driver go to Settings - Login Items & Extensions - Driver Extensions and enable the new CH34x Driver - REBOOT ![Screenshot 2024-12-15 at 12 19 51 PM](https://private-user-images.githubusercontent.com/56480819/395880284-a980fdce-02e1-4136-9723-9b93e7b211fe.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQyOTk3OTUsIm5iZiI6MTczNDI5OTQ5NSwicGF0aCI6Ii81NjQ4MDgxOS8zOTU4ODAyODQtYTk4MGZkY2UtMDJlMS00MTM2LTk3MjMtOWI5M2U3YjIxMWZlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMTUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjE1VDIxNTEzNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTc5NmIzNjMyMzIxYmY5Mjk3M2Q0MDFhNzJjY2YyMjdjZjM3NjA0ZTkxMDZmMDBhOTk5MjE0MWJmYTUzMWFjNzUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.amigcucOygdB6Czxdko8lPssU90UV6ZZ91JP-EZvhCc)
> 
> Thorny Instructions: Open Thorny - go to the RUN Menu Select Configure Interpreter ![Screenshot 2024-12-15 at 12 23 47 PM](https://private-user-images.githubusercontent.com/56480819/395880306-cee2bd7d-99bf-4fc1-a274-3330feb32ca3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQyOTk3OTUsIm5iZiI6MTczNDI5OTQ5NSwicGF0aCI6Ii81NjQ4MDgxOS8zOTU4ODAzMDYtY2VlMmJkN2QtOTliZi00ZmMxLWEyNzQtMzMzMGZlYjMyY2EzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMTUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjE1VDIxNTEzNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTFmYWYxYzIxODYzMTJhOTdjMTAxZTA1MzE0ZmQ2NjIxMzVlYmZiYTI2N2UzOThkYTQzZjc0MDg4OGRmNmM5NGEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.m2FzgArQ1CoYQ5AjyG84aJsCVBZV62t_82xOu1K8GhI)
> 
> Follow the Rest of the instructions

Hi,
Can you reset the panda touch now?
There is document you can check it out also.

https://github.com/bigtreetech-beta/PandaTouch-Support/blob/main/Recovery_toolV1.0.6/flash_on_macos.doc

closed 2024-12-17 by dilixi