bigtreetech/PandaTouch #203: Thumbnail no longer visible after 1.0.6.1 update

Neuropsihopuppu, 2024-12-18:
Thumbnail no longer visible while printing  after  updating to 1.0.6.1 ,have not changed anything else but i only see btt logo now

koffienl, 2024-12-19:
Thumbnails seem to be really buggy. When powering on the panada touch while printing you don't get thumbs at all.
After that you will get thumbs, but I never got a thumbnail for the current printing job if it is not reprint. When doing a reprint I get the thumb.

Neuropsihopuppu, 2024-12-19:
They worked just fine for me untill the last update from 1.0.6 to 1.0.6.1

dilixi, 2024-12-19:
> Thumbnail no longer visible while printing after updating to 1.0.6.1 ,have not changed anything else but i only see btt logo now

We did change something about loading thumbnail which panda touch doesn't load thumbnails via ftps when the printer is printing.

Neuropsihopuppu, 2024-12-19:
Could you please explain a little more because that does not make much sense to me? So what you are saying is that if i send a print from bambu studio i will no longers see thumbnails?

koffienl, 2024-12-19:
> They worked just fine for me untill the last update from 1.0.6 to 1.0.6.1

I just got it lost week, so I immetially go from version old to 1.0.6.1 without ever using 1.0.6.0
Perhaps I should downgrade to 1.0.6.0

dilixi, 2024-12-19:
> Could you please explain a little more because that does not make much sense to me? So what you are saying is that if i send a print from bambu studio i will no longers see thumbnails?

Some people said that sometimes the printer will reboots just when the panda touch is reconnecting to it, so we need to stop rebooting when the printer is printing.

We will find a real reason and solve this problem in a better way.

dilixi, 2024-12-19:
If you need a version which the panda touch will loads thumbnail even when the printer is printing then you just need to send me a email about this  thanks.
1284513393@qq.com

Neuropsihopuppu, 2024-12-19:
Will this get fixed in a future update of the firmware? And is it posible for me to just downgrade to 1.0.6 myself?

dilixi, 2024-12-19:
> Will this get fixed in a future update of the firmware? And is it posible for me to just downgrade to 1.0.6 myself?

Don't do that downgrade,please send me a email.

Neuropsihopuppu, 2024-12-19:
Ok

ChristianBracket, 2025-04-19:
Same for me on the new firmware 1.0.7