bigtreetech/PandaTouch #209: 1.0.6 Panda PWR

Wizzard1313, 2024-12-23:
I updated to 1.0.6.2 and my Panda PWR tab disappeared. I can get it back downgrading to 1.0.5

dilixi, 2024-12-23:
> I updated to 1.0.6.2 and my Panda PWR tab disappeared. I can get it back downgrading to 1.0.5

Hi,
There is a option to enable/disable it in Panda Touch settings.

Wizzard1313, 2024-12-23:
I found the setting. In my opinion it should not default to off

closed 2024-12-23 by Wizzard1313

dilixi, 2024-12-23:
> I found the setting. In my opinion it should not default to off

Thanks,maybe We need change some logic.

bigtreetech, 2024-12-23:
> I found the setting. In my opinion it should not default to off

So many users complained that the tab existed when they were not using the Panda PWR. We will have to investigate better logic.