bigtreetech/PandaTouch #215: Printing from SD Card

TanoAvl, 2025-01-02:
After Updating to Firmware 1.0.6.3 with the German image I am not able to start a print from the SD Card by the Panda Touch. I have to start the print by Bambu studio or Handy.

dilixi, 2025-01-03:
> After Updating to Firmware 1.0.6.3 with the German image I am not able to start a print from the SD Card by the Panda Touch. I have to start the print by Bambu studio or Handy or by the the original display.

Hi,
Do you login an account on panda touch?

Please send me an email,thanks.

1284513393@qq.com

TanoAvl, 2025-01-03:
Yes my Account is logged in.

bigtreetech, 2025-01-03:
> Yes my Account is logged in.

Did you log out and back in after updating?

TanoAvl, 2025-01-03:
I’m not sure I will try it.

closed 2025-01-03 by TanoAvl