bigtreetech/PandaTouch #216: Change of nozzle temperature

Janne-Swe, 2025-01-04:
Can't change nozzle temperature. Running V 1.0.6.3 and English version.

dilixi, 2025-01-04:
> Can't change nozzle temperature. Running V 1.0.6.3 and English version.

Do you login an account on panda touch and enable cloud mode in printer's edit page on panda touch?

Janne-Swe, 2025-01-04:
No, I can try that. But I can control the light and movement.But I will try what you propose.Skickat från min iPhone4 jan.

bigtreetech, 2025-01-04:
You must log in as stated in the manual and wiki.

closed 2025-01-04 by bigtreetech

Janne-Swe, 2025-01-04:
Hi,I have now tried to log in by put in my mail adress and then send a code.It start to try and after some seconds it fail. It complain on some networkissue. But I have good wifi connection.Thank’s a lot for your support.Skickat från min iPad4 jan.

Janne-Swe, 2025-01-04:
Hi again,I found the problem. Was some issue with my mail in the Bambu account, now solved.Thanks for your support.Skickat från min iPad4 jan.