bigtreetech/PandaTouch #218: Filament in Panda not the same as in mobile and PC

Duffy987, 2025-01-05:
Hello everyone,

I have the problem that the filament created in OrcaSlicher (PC) is not available in Panda.

For example, Giantarm and Sunlu are available in the Orca Slicer:
[image]

In the mobile phone too:
[image]

But unfortunately not in Panda Touch:
[image]

Is there a way to display the created filaments in Panda?

Regards Duffy

dilixi, 2025-01-06:
> Hello everyone,
> 
> I have the problem that the filament created in OrcaSlicher (PC) is not available in Panda.
> 
> For example, Giantarm and Sunlu are available in the Orca Slicer: ![grafik](https://private-user-images.githubusercontent.com/120317770/400193481-10c59f88-1f44-4d61-93f3-e405fa3a9021.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzYxNDg5MDcsIm5iZiI6MTczNjE0ODYwNywicGF0aCI6Ii8xMjAzMTc3NzAvNDAwMTkzNDgxLTEwYzU5Zjg4LTFmNDQtNGQ2MS05M2YzLWU0MDVmYTNhOTAyMS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMTA2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDEwNlQwNzMwMDdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00NGE0MjA5NDNmZTMzOWEwNTI0MGM3MDc0OGE4MGVlMTM4Y2Y4NjU0OWY2YzJjNWZkMTc1OGE0ZjcyMTIxZDkwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.3ZYXe0G2P8sgDiyYPiSVR02TrQUVhPtw1ZMrlgrCRsQ)
> 
> In the mobile phone too: ![Handy](https://private-user-images.githubusercontent.com/120317770/400193546-2d2a7005-79c7-4c2c-8a6e-156c4d250bfd.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzYxNDg5MDcsIm5iZiI6MTczNjE0ODYwNywicGF0aCI6Ii8xMjAzMTc3NzAvNDAwMTkzNTQ2LTJkMmE3MDA1LTc5YzctNGMyYy04YTZlLTE1NmM0ZDI1MGJmZC5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMTA2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDEwNlQwNzMwMDdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03NjU1NjI5NDdlNWEwZWQ5OGY2MWMyODc2NzkyZDYyN2VjMDE1NzBmOWFmZGYwYTAwNTEyMjRkMGQyZTBjYzYzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.mkIYH9jXS49ItjvvkqIDQX5GZL3KQbSPGF_wld4h4hc)
> 
> But unfortunately not in Panda Touch: ![Pamda](https://private-user-images.githubusercontent.com/120317770/400193581-9b703de5-9760-4a73-a2fa-c7e6a46ebb59.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzYxNDg5MDcsIm5iZiI6MTczNjE0ODYwNywicGF0aCI6Ii8xMjAzMTc3NzAvNDAwMTkzNTgxLTliNzAzZGU1LTk3NjAtNGE3My1hMmZhLWM3ZTZhNDZlYmI1OS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMTA2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDEwNlQwNzMwMDdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1kODQ1NmZkOGNjMmU1M2I3YWZhYzUxMTAxOGFmZmVlODZjZTc5Y2FlZjVkMjhkMWM3ZGUyNGJiNTI1ZDczNGQ1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.conAaf4_9uH9p4TPBtktHM1NuO_XtC4jsQpgK8ykQnw)
> 
> Is there a way to display the created filaments in Panda?
> 
> Regards Duffy

Hi,
We will add this feature in new firmware.

zeemyself, 2025-01-08:
So you found a solution to work with custom filament profile? [!107 ](https://github.com/bigtreetech/PandaTouch/issues/107) cause last time I reported panda touch doesn't have resources

bigtreetech, 2025-01-08:
> So you found a solution to work with custom filament profile? [!107 ](https://github.com/bigtreetech/PandaTouch/issues/107) cause last time I reported panda touch doesn't have resources

Yes, we found some clever tricks to completely stretch the hardware to the limits and we are using them as much as possible.

NefariousAryq, 2025-04-12:
I just purchased a Panda Touch not realizing that I couldn't control custom filaments with it. Is there a timetable for when this feature may be coming, @bigtreetech? May have to return the Panda Touch unless it's coming at some point soon, as filament management is the biggest reason I purchased the PT. Thanks!

keithwkx, 2025-04-14:
@bigtreetech 回复一下呗

raihei, 2025-04-14:
@dilixi Do you mean if we could do testing of a new version? Yes, I can do that. If working on the filaments anyway, maybe you can also check #269

mytechguyri, 2025-04-15:
> I just purchased a Panda Touch not realizing that I couldn't control custom filaments with it. Is there a timetable for when this feature may be coming, @bigtreetech? May have to return the Panda [...]

Honestly the printer and slicer only care about one thing...filament type...is it PLA or PETG or PA6-CF etc... and color... That's what the slicer uses to match the custom filament you have in the slicer to the filament in the AMS...i.e. if you have Generic PLA red in the PandaTouch screen, and you have Overture Rapid PLA red as a custom filament profile in the slicer and are using that in your model, then it will see PLA and Red and properly map that slot, and the slicer will use the custom filament profile you specified

dilixi, 2025-04-24:
@raihei 
@keithwkx 
@NefariousAryq 
Hi all,
I have added this feature now.

Please have a try:
https://github.com/bigtreetech-beta/PandaTouch-Support/blob/main/firmware/v1.0.7.2_beta1.bin

keithwkx, 2025-04-24:
> [@raihei](https://github.com/raihei) [@keithwkx](https://github.com/keithwkx) [@NefariousAryq](https://github.com/NefariousAryq) Hi all, I have added this feature now.
> 
> Please have a try: https://github.com/bigtreetech-beta/PandaTouch-Support/blob/main/firmware/v1.0.7.2_beta1.bin

got it will flash and try shortly. keep u posted

raihei, 2025-04-24:
@dilixi Which feature exactly? I installed the new firmware and still have the same filament selection available.

@mytechguyri That is not 100% correct, if you work with calibration profiles. For these profiles (which are stored in the printer), the selected filament has to match exactly the type which was used to do the calibration.

For example, if you did the calibration with "SUNLU PETG" which has the filament-id "GFSNL08" and you set the filament on the Panda Touch to "Generic PETG" (or in the slicer software) which has the filament-id "GFG99" the calibration profile cannot be used.

dilixi, 2025-04-24:
> [@dilixi](https://github.com/dilixi) Which feature exactly? I installed the new firmware and still have the same filament selection available.
>
> [@mytechguyri](https://github.com/mytechguyri) That is [...]

Panda Touch doesn't support custom filament before.
And Panda Touch supports it now.

keithwkx, 2025-04-24:
@dilixi working nicely. may I ask these profile syncs via cloud or? As I noticed I might have some duplicate profiles in the dropdown

[image]

dilixi, 2025-04-24:
> [@dilixi](https://github.com/dilixi) working nicely. may I ask these profile syncs via cloud or? As I noticed I might have some duplicate profiles in the dropdown, so would like to find a way to modify or mirror them correctly to my pc
> 
> [image] [image]

From cloud.

https://github.com/greghesp/ha-bambulab/issues/540#issuecomment-2186383233

dilixi, 2025-04-24:
> > > [@dilixi](https://github.com/dilixi) working nicely. may I ask these profile syncs via cloud or? As I noticed I might have some duplicate profiles in the dropdown, so would like to find a way to modify or mirror them correctly to my pc
> > > [image] ![Image](https://private-user-images.githubusercontent.com/28388406/437000917-255181fe-4085-4455-8bab-e3995be212af.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDU0OTYyNzYsIm5iZiI6MTc0NTQ5NTk3NiwicGF0aCI6Ii8yODM4ODQwNi80MzcwMDA5MTctMjU1MTgxZmUtNDA4NS00NDU1LThiYWItZTM5OTViZTIxMmFmLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA0MjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNDI0VDExNTkzNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWJkM2NhYjdjODRhZDJhZWM4N2YzZGFlMWZkM2Q4ODZiZGM3NjM2ZWQzODJkZTA1Mzk5ZjNlNzQxNmFlYmNjMDQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.gfFMM0zzmbWa53CENK_ArenqVcxR6FJkblpaDfUYqcA)
> > 
> > 
> > From cloud.
> > [greghesp/ha-bambulab#540 (comment)](https://github.com/greghesp/ha-bambulab/issues/540#issuecomment-2186383233)
> 
> I see so in other words, I need to remove the profile from my pc, let the cloud sync. Then the changes will reflect on panda touch. Correct?

你会用中文吗？
不知道你说的重复的具体是什么情况，你可以试一下，电脑要联网。

keithwkx, 2025-04-24:
> 你会用中文吗？ 不知道你说的重复的具体是什么情况，你可以试一下，电脑要联网。

我重复是指，我之前用“点维”创建了个耗材资料，后期改用英文"Dowell"了。你看我的图里，两个耗材都出现，间接性导致重复。就像了解是否有方法移除“点维”这个耗材资料

dilixi, 2025-04-24:
> > > > > [@dilixi](https://github.com/dilixi) working nicely. may I ask these profile syncs via cloud or? As I noticed I might have some duplicate profiles in the dropdown, so would like to find a way to modify or mirror them correctly to my pc
> > > > > [image] ![Image](https://private-user-images.githubusercontent.com/28388406/437000917-255181fe-4085-4455-8bab-e3995be212af.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDU0OTYyNzYsIm5iZiI6MTc0NTQ5NTk3NiwicGF0aCI6Ii8yODM4ODQwNi80MzcwMDA5MTctMjU1MTgxZmUtNDA4NS00NDU1LThiYWItZTM5OTViZTIxMmFmLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA0MjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNDI0VDExNTkzNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWJkM2NhYjdjODRhZDJhZWM4N2YzZGFlMWZkM2Q4ODZiZGM3NjM2ZWQzODJkZTA1Mzk5ZjNlNzQxNmFlYmNjMDQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.gfFMM0zzmbWa53CENK_ArenqVcxR6FJkblpaDfUYqcA)
> > > > 
> > > > 
> > > > From cloud.
> > > > [greghesp/ha-bambulab#540 (comment)](https://github.com/greghesp/ha-bambulab/issues/540#issuecomment-2186383233)
> > > 
> > > 
> > > I see so in other words, I need to remove the profile from my pc, let the cloud sync. Then the changes will reflect on panda touch. Correct?
> > 
> > 
> > 你会用中文吗？ 不知道你说的重复的具体是什么情况，你可以试一下，电脑要联网。
> 
> 我重复是指，我之前用“点维”创建了个耗材资料，后期改用英文"Dowell"了。你看我的图里，两个耗材都出现，间接性导致重复。现在想了解是否有方法移除“点维”这个耗材资料。
> 
> 据我知，这是从云端调来的资料。那我是不是得删除pc端的耗材资料让云端同步即可？

在电脑端删除后是可以同步到云端的，保持电脑网络畅通。

dilixi, 2025-04-24:
> 我重复是指，我之前用“点维”创建了个耗材资料，后期改用英文"Dowell"了。你看我的图里，两个耗材都出现，间接性导致重复。现在想了解是否有方法移除“点维”这个耗材资料。
>
> 据我知，这是从云端调来的资料。那我是不是得删除pc端的耗材资料让云端同步即可？

这里有详细的方法
https://wiki.bambulab.com/zh/bambu-studio/create-filament

raihei, 2025-04-24:
Do you mean the mobile app Bambu Handy? I don't have it, because it only works in cloud mode. So I cannot confirm.
But according to the creator of this thread they are available, see first post.

> Hello everyone,
>
> I have the problem that the filament created in OrcaSlicher (PC) is not available in Panda.
>
> For example, Giantarm and Sunlu are available in the Orca Slicer: [image]
>
> In the mobile [...]

Here are the filament codes, taken from BambuStudio sources:

SUNLU Wood GFSNL07
SUNLU Silk GFSNL05
SUNLU PLA+ GFSNL03
SUNLU PLA+ 2.0 GFSNL04
SUNLU Matte GFSNL02
SUNLU Marble GFSNL06
SUNLU PETG GFSNL08

NefariousAryq, 2025-04-24:
Ah, yeah, if this solution requires the Bambu cloud then it's a non-starter for me unfortunately. I've put my printer in LAN mode due to the upcoming restrictions.

dilixi, 2025-04-25:
> Same for me. I don't know if it wise from BTT to rely so much on the Bambu cloud. If Bambu locks out all 3rd party suppliers in the future, a lot of the product features will be broken. Better way would be, to implement functions, that also work in LAN mode.

Maybe we can add some of them because there are too many public filaments.

dilixi, 2025-04-25:
Maybe we can add some of them because there are too many public filaments.

[public filaments.json](https://github.com/user-attachments/files/19902323/public.filaments.json)

NefariousAryq, 2025-05-01:
I have unfortunately reached the end of my return period, so I've returned my Panda Touch. I'd be open to purchasing again in the future if a solution to this can be found that doesn't require use of Bambu's cloud services. Thanks for your assistance in this matter.

keithwkx, 2025-05-01:
> Maybe we can add some of them because there are too many public filaments.
> 
> [public filaments.json](https://github.com/user-attachments/files/19902323/public.filaments.json)

how do i use this .json?
And a wild thought. Why not a public repo where users can submit their filament brand and type.
Then u can use it