bigtreetech/PandaTouch #224: Panda Touch Crashing when Docked

phil-the-dev, 2025-01-15:
I noticed this on 1.0.6.1 and I've upgraded to 1.0.6.3 and still have the issue. I did a full factory reset and the issue is still happening.

In short, when I dock the Touch, the device seems to crash consistently. I've had the device for 6-7 months with no problems and I think it was the 1.0.6 update that caused this to start.

Here's a video:
<video src="https://github.com/user-attachments/assets/2fc64567-2cfb-4041-8ed7-151ade8d6cba" />

looxonline, 2025-01-15:
Does it stop if you leave it docked for a while? It may be that your USB port on the printer is unable to deliver the current needed by the touch and the battery charging circuit so the touch is resetting. Leaving it docked for a while may charge the battery back up. Have you tried another cable?

bpivk, 2025-01-15:
I have the same problem. It doesn't stop if you have a full battery and it happens randomly.

phil-the-dev, 2025-01-16:
@looxonline I unplugged and replugged the wire and it's now working flawlessly! I must have knocked the cable loose some how. Thank you for you help!

closed 2025-01-16 by phil-the-dev