bigtreetech/PandaTouch #225: Firmware Update Introducing New Authorization Control System #36

koffienl, 2025-01-16:
Any idea if this will block the use of the panda touch?

https://blog.bambulab.com/firmware-update-introducing-new-authorization-control-system-2/

Launching first for X Series printers, with P and A Series updates planned for future release

As part of our ongoing commitment to enhance the overall security of our products, we are introducing an authorization and authentication protection mechanism for the connection and control of Bambu Lab 3D printers. This step is a significant security enhancement to ensure only authorized access and operations are permitted.

Critical Operations That Require Authorization
The following printer operations will require authorization controls:

Binding and unbinding the printer.
Initiating remote video access.
Performing firmware upgrades.
Initiating a print job (via LAN or cloud mode).
Controlling motion system, temperature, fans, AMS settings, calibrations, etc.

mytechguyri, 2025-01-18:
I expect that's the whole point of Bambu issuing this firmware update... If I can save $500 and get a P1S instead of an X1C and get a touch screen for $59 I'm sure Bambu isnt happy about that... It damn sure isn't about "security" like they claim, since third party apps and devices can't control the printer without the printer access code, which is only available on the printer screen, and requires you to type it into the third party devices or app, specifically granting it access.

philchfree, 2025-01-18:
In the wiki, it's made clear we should not update to this firmware:

Will aftermarket accessories stop working? 

If aftermarket accessories, such as Panda Touch, control features like binding/unbinding the printer, initiating print jobs, axis movement, or calibrations, these functions will no longer work. However, other features, such as monitoring printer status (e.g., temperature, position, speed) and LED control, will remain accessible.

Dad-Ju, 2025-01-20:
Seams like there will be a Dev mode supporting MQTT and FTP
https://forum.bambulab.com/t/updates-and-third-party-integration-with-bambu-connect/137408

```
Developer Mode (Optional): For advanced users of the X1, P1, A1, and A1 Mini who prefer full control over their network security, an option will be available to leave the MQTT channel, live stream, and FTP open. This feature must be manually enabled on the printer, and users who select this option will assume full responsibility for securing their local network environment. Please note that Bambu Lab will not be able to provide customer support for this mode, as the communication protocols are not officially supported.
```