bigtreetech/PandaTouch #228: PandaTouch restarting when waking from sleep

TaleSter, 2025-01-20:
not sure if im the only one with this issue, but its annoying because it takes extra time for it to reconnect to the wifi and printer

https://github.com/user-attachments/assets/be7533b5-4b7b-4d04-b9ea-abdecc90bdaa

lw348653864, 2025-01-22:
I have the same problem.

dilixi, 2025-01-23:
@TaleSter 
@lw348653864 

Hi all,
Please have a try with the following steps:
1 Switch the power mode to battery. 切换到电池档位。
2 Move the Panda Touch on a table or another place. 把Panda Touch放在另外一个地方。
3 Chage the time to 2 minutes.  把休眠时间改为2分钟。
4 Click the screen after it is sleeping. 休眠时点一下屏幕。

bpivk, 2025-01-28:
Same problem here. Dc mode connected to the printer with the included cable. The screen works fine but when I click on the screen after sleep I get a crash and device reboots.

TaleSter, 2025-02-03:
> [@TaleSter](https://github.com/TaleSter) [@lw348653864](https://github.com/lw348653864)
>
> Hi all, Please have a try with the following steps: 1 Switch the power mode to battery. 切换到电池档位。 2 Move the [...]

issue still occurs

bpivk, 2025-02-08:
I have changed the cable and connected the device to Panda Power. Not reboots.

I would suggest against telling customers to plug it into the printer or warning them about it as it seems that the printer does not provide enough power resulting in boot loops. 

So far the only change I did was change the stock cable that came with Touch with the stock cable that came with Power and plugging it there. But it does look like that the problem was with the main board not providing enough power.

Archer032, 2025-02-21:
I still have the same issue even though i updated the firmware to 1.0.6.3 and it's sporadic, but still very annoying.  It seems that in battery mode it's working for now.
BTW if the screen is in battery mode, does it charge while plugged in to a charger?

yhh661, 2025-02-25:
Happens to me either, when the screen is in dc mode. Changing of USB cable doesn't help