bigtreetech/PandaTouch #231: Z axis direction backwards

K2AAE, 2025-01-21:
I ran into some blob issues and while trouble shooting I tried to raise (Move the tool head up, away from the bed) using the arrows at the top of the Z axis display BUT it sent the nozzle into the bed with my new, hardly used Bambu Lab Carbon Smooth plate. The A1 Mini control arrow that points up, is at the top of the icon, moves the head up, away from the plate, BTT Panda-Touch move the head down into the plate.
Why? 
On what Planet does Up mean Down?  I finally figured out the two arrows pointing at each other with a number to the right is suppose to mean the distance of the move but I have to ask, on what planet does a user not know a button with 10 means 10mm step if you click/press it? 

If you have a button to move something please place a direction of the move on the button!
 Normally if I see one button above another button I would think that means the top button move something up, I understand where this was created is on the other side of the planet, is this just a crude joke on us (North America)?

dilixi, 2025-01-22:
> I ran into some blob issues and while trouble shooting I tried to raise (Move the tool head up, away from the bed) using the arrows at the top of the Z axis display BUT it sent the nozzle into the [...]

Hi,
It is a bug not a crude joke.
Because BBL just change the logic of z-axis in the new firmware of A1 Mini.

dilixi, 2025-03-22:
this bug has been fixed

closed 2025-03-22 by dilixi