bigtreetech/PandaTouch #233: Files on USB drive not loading.

JustYourAverageGithubUser, 2025-01-23:
I have a 32GB PNY USB 3.0 flash drive formatted in FAT32 with the name "pandatouch". When I connect it to the Panda Touch it recognizes the USB is connected, but doesn't see any files or folders. I'm uploading .3mf files sliced in OrcaSlicer on Plate #1 using the "export plate sliced file" and I'm on firmware version 1.0.6.3

looxonline, 2025-01-23:
IIRC the max drive size is 8GB. Could you try a smaller drive?

JustYourAverageGithubUser, 2025-01-24:
Sorry for the late reply. I was wrong in my original post, it was a 64GB drive. I tried using the microSD card through a USB reader and that works, the microSD that my P1S is 32GB, so I think that's the size limit. Switching to an 8GB Flash drive I bought works. It took me awhile cause I only had 64GB flash drives around.

closed 2025-01-24 by JustYourAverageGithubUser