bigtreetech/PandaTouch #236: Bug: files uploaded in LAN mode not visible in file list

AlvaroBrey, 2025-01-27:
Sort of related to https://github.com/bigtreetech/PandaTouch/issues/229, but not the same.

Files uploaded in LAN mode are not visible in the file list, so the print can't be triggered from there. 

I _think_ this may be because files from cloud are like `file.3mf`, but LAN mode files are `file.gcode.3mf` and the Touch may be doing some wrong filtering?

The LAN files are viewable and printable in the stock display.

zeemyself, 2025-01-28:
[image]
I think you need to change directory to root instead of cache
root=lan
cache=online

AlvaroBrey, 2025-01-28:
You are absolutely right, thanks! That was unexpected as in my FTP client they show up inside /cache 🤷

closed 2025-01-28 by AlvaroBrey