bigtreetech/PandaTouch #237: Feature request: Skip object

mike391, 2025-02-07:
In preparation for any cloud mode shenanigans, the homeassistant bambu lab integration team figured a way to implement skip object without using the cloud.  It seems fairly simple and doesn't require reslicing or anything cpu intensive. This should be easy to port into panda touch  as the logic is already done here:
https://github.com/greghesp/ha-bambulab/issues/866

bigtreetech, 2025-02-07:
Thanks for the ping. We have already been working on this for some time. It will be coming in a future update.

bigtreetech, 2025-02-07:
@dilixi

dilixi, 2025-02-08:
> In preparation for any cloud mode shenanigans, the homeassistant bambu lab integration team figured a way to implement skip object without using the cloud. It seems fairly simple and doesn't require reslicing or anything cpu intensive. This should be easy to port into panda touch as the logic is already done here: [greghesp/ha-bambulab#866](https://github.com/greghesp/ha-bambulab/issues/866)

Thanks.
That helps me a lot.

dilixi, 2025-03-22:
v1.0.7 has this feature

closed 2025-03-22 by dilixi

raihei, 2025-04-17:
@dilixi What are the preconditions so an object can be skipped? My Panda touch is on V1.0.7 but I don't have the option to skip. I have 5 objects on the plate and the print sequence is "by object". The print was started via OrcaSlicer, printer is in LAN mode. Is that an unsupported combination?

dilixi, 2025-04-17:
> [@dilixi](https://github.com/dilixi) What are the preconditions so an object can be skipped? My Panda touch is on V1.0.7 but I don't have the option to skip. I have 5 objects on the plate and the [...]

Yes.
You need to login an account on panda touch and enable cloud mode.

raihei, 2025-04-17:
@dilixi thanks for your answer.

But this feature request was about skip object support without cloud and you confirmed, that it was implemented in V1.0.7?

> In preparation for any cloud mode shenanigans, the homeassistant bambu lab integration team figured a way to implement skip object without using the cloud. It seems fairly simple and doesn't require [...]

dilixi, 2025-04-18:
> [@dilixi](https://github.com/dilixi) thanks for your answer.
>
> But this feature request was about skip object support without cloud and you confirmed, that it was implemented in V1.0.7:

Yes.
This requirement has not been fully met yet. 
Panda touch needs to download whole .3mf file via ftps, but there is not enough ram to save this file and also it needs ram to unzip the .3mf and get the png file from it.
We are trying to find a way to solve this problem.

raihei, 2025-04-20:
Okay, thank you for the explanation!

mike391, 2025-04-20:
> Yes. This requirement has not been fully met yet. Panda touch needs to download whole .3mf file via ftps, but there is not enough ram to save this file and also it needs ram to unzip the .3mf and [...]

I asked about a similar thing a month after release. The general explanation was that to address this ram issue, BTT was making a processing unit expansion that was capable of bridging this gap. Is this no longer being developed?

BradCOnGitHub, 2025-05-25:
This should be reopened. We need something to track this issue until it actually works without cloud mode.