bigtreetech/PandaTouch #247: Model image glitch?

maple-underscore, 2025-02-27:
So... yeah. Coming back to the print and greeted with this mess

[image]

[image]

[image]

[image]

Can anybody tell me what this is?
Model link: https://makerworld.com/en/models/432500#profileId-555554
The 0.16mm, 44 min one
I use printed two-color filament feeding from the external holder, printed with PLA Basic and set to Bambu PLA Basic Light Grey

looxonline, 2025-02-27:
Probably just some sort of corruption in the download from the FTP. This is the first report of it happening so it is definitely not systemic. Will leave it open in the hopes that if others see the same thing they will come here and report it too. If nothing else is reported in a month then we can attribute it to a random download glitch and close it @BTTJUSTIN

maple-underscore, 2025-02-27:
I will try printing it again this evening to see if the issue arises.

BTTJUSTIN, 2025-02-28:
ok, I will mark it down