bigtreetech/PandaTouch #249: SD card browsing shows no year in date row

raihei, 2025-03-04:
When browsing the printers SD card via the panda touch, no year is shown in the "Date" row, only "----"

[image]

Printer: P1S, V1.07.00.00
Touch: V1.0.6.3

Printer is in "LAN only" mode, Panda Touch is also not connected to the cloud.

When I browse the SD card via SFTP, the dates are shown:

[image]

What can be the cause?

dilixi, 2025-03-05:
> When browsing the printers SD card via the panda touch, no year is shown in the "Date" row, only "----"
>
> [image]
>
> Printer: P1S, V1.07.00.00 Touch: V1.0.6.3
>
> Printer is in "LAN only" mode, Panda [...]

Hi,
Panda Touch does not support the 'RTC' currently, so it doesn't know what the day is today.
Panda Touch will add this feature in the new firmware(not V1.0.7).

zeemyself, 2025-03-06:
I don't think RTC support(to get today date) is the problem, it is not getting year to able to sort file properly

ex. all files (YYYY-MM-DD)
2025-03-06
2024-09-02
2024-02-10

this will be show in panda as
----/02/02
----/03/06
----/09/02

which is completely wrong because it not get year in consideration(sort)

dilixi, 2025-03-06:
> I don't think RTC support(to get today date) is the problem, it is not getting year to able to sort file properly
>
> ex. all files (YYYY-MM-DD) 2025-03-06 2024-09-02 2024-02-10
>
> this will be show in [...]

Could you please show me a photo of panda touch which contains different years(not 2025)?
'----'  means 2025(when this year is 2025).

If the data of file is '2024'，it should not be '----'.

dilixi, 2025-03-07:
> I don't think RTC support(to get today date) is the problem, it is not getting year to able to sort file properly
>
> ex. all files (YYYY-MM-DD) 2025-03-06 2024-09-02 2024-02-10
>
> this will be show in [...]

yes, you are right.
Panda Touch is using the command "LIST" which can not get the year.

line:-rw-rw-rw-   1 root  root   1629155 Jan 03 12:06 5065_flywheel_plate_1.gcode
line:-rw-rw-rw-   1 root  root     41078 Jan 09 09:26 Calibration_cube_XYZ.3mf

We will add this feature in the new firmware.

dilixi, 2025-04-18:
> I don't think RTC support(to get today date) is the problem, it is not getting year to able to sort file properly
>
> ex. all files (YYYY-MM-DD) 2025-03-06 2024-09-02 2024-02-10
>
> this will be show in [...]

Hi all,
It should can work now.

[image]

please have a try.
https://github.com/bigtreetech-beta/PandaTouch-Support/blob/main/firmware/v1.0.7.1_beta6.bin

raihei, 2025-04-18:
Yes, seems to work now, thanks. On older directories, I can also see, that years other than 2025 also work.

[image]

When testing I encountered, that the print dialog was changed. Longer filament names now don't fit anymore:

[image]

dilixi, 2025-04-18:
> Yes, seems to work now, thanks. On older directories, I can also see, that years other than 2025 also work.
> 
> [image]
> 
> When testing I encountered, that the print dialog was changed. Longer filament names now don't fit anymore:
> 
> [image]
> 
> Edit: Nevermind, I just saw, that you can actually scroll in that filament field 😮👍

Ok.
I will fix this next Monday.

bigtreetech, 2025-04-18:
> Ok. I will fix this next Monday.

Don't worry. It already works. He edited the response.

raihei, 2025-04-18:
Yes, it certainly works, but it doesn't look very nice. 😄Maybe it is possible to make the filament rectangles wider, there should be enough space in the dialog window.

closed 2025-04-24 by dilixi