bigtreetech/PandaTouch #252: When loading a file from the panda touch it defaults to the A1 slot even if I specify another slot in the AMS

MGaming7378, 2025-03-17:
Ever since I last updated my Panda Touch display, whenever I attempt to load a print from the USB port it defaults to the A1 slot in the AMS even if I specify any other slot it defaults to the A1 slot and starts printing with that filament instead of the one selected, even if I manually change it it automatically changes back to the A1 slot and continues printing. Never really used it to start a print because of this issue because I have some prints that I only saved to the USB drive on the Panda Touch.

TargiX, 2025-03-18:
Yeah, same issue. Not working at all.

dilixi, 2025-03-18:
> Ever since I last updated my Panda Touch display, whenever I attempt to load a print from the USB port it defaults to the A1 slot in the AMS even if I specify any other slot it defaults to the A1 [...]

Hi,
What's the firmare version of AMS?
Which region did you select when you config the network of printer?

Can you send me more about this issue via email?
My email is 1284513393@qq.com.

dilixi, 2025-04-11:
@TargiX
@MGaming7378

Hi all,
Do you have time to help me to get some logs?