bigtreetech/PandaTouch #253: Reprint feature is not working

TargiX, 2025-03-18:
On my P1S when I hit 'Reprint" button it shows the menu to initialize print, when I accept it just refreshes to the home screen and nothing happens. Anybody have same issues, any solutions?

PsekTekk, 2025-03-31:
Yes, i have same problem on P1S panda touch v1.07. Touch print from history. Printer still heating bed and waiting on AMS and than nothing. From mobile is not problem with print.

dilixi, 2025-03-31:
> Yes, i have same problem on P1S panda touch v1.07. Touch print from history. Printer still heating bed and waiting on AMS and than nothing. From mobile is not problem with print.

Hi,
Did you reprint it from history panel?
Did it ask you to select slot ?

PsekTekk, 2025-04-01:
Did you reprint it from history panel?  Yes, from history panel.

Did it ask you to select slot ?  No, I didn't see anything. And should I?

dilixi, 2025-04-01:
> Did you reprint it from history panel? Yes, from history panel.
> 
> Did it ask you to select slot ? No, I didn't see anything. And should I?

The print history was generated from cloud and sometime it can not be reprinted.
Can I login your account to test It?

Where did you send the print task from handy app? Is main panel or history panel?

Was it a same history record which can be reprint by handy app when you send from your handy?

PsekTekk, 2025-04-02:
> The print history was generated from cloud and sometime it can not be reprinted. Can I login your account to test It?
>
> Where did you send the print task from handy app? Is main panel or history [...]

No, I didn´t send the print task from handy app, I sent it from the Panda touch, panel "print history".I never printed from handy app before, only to test it. 

Yes, the same history record can be reprinted from handy. 

You can login to my account, where can I send you my login details?

dilixi, 2025-04-02:
> No, I didn´t send the print task from handy app, I sent it from the Panda touch, panel "print history".I never printed from handy app before, only to test it.
>
> Yes, the same history record can be [...]

> Yes, the same history record can be reprinted from handy.
where did you reprint it? From handy main page? Or handy history panel?

>You can login to my account, where can I send you my login details?
You can send me a message to1284513393@qq.com and I will send you the details.