bigtreetech/PandaTouch #256: cant find yml eng file 1.0.7.x  to translate.

Winky1st, 2025-03-23:
Hi, where can I find the necessary files (eng.yml) version 1.0.7.x to update a translation?

also https://pandatouch.us.kg/ is still offline.

previous translation was posted here. https://github.com/bigtreetech/PandaTouch/issues/82#issuecomment-2629741076

I saw that the argument was closed and @dilixi kindly posted this tool : https://github.com/bigtreetech-beta/PandaTouch_Addtion_IMG
are there any instructions available?
thanks and cheers all

dilixi, 2025-03-24:
> Hi, where can I find the necessary files (eng.yml) version 1.0.7.x to update a translation?
> 
> also https://pandatouch.us.kg/ is still offline.
> 
> previous translation was posted here. [#82 (comment)](https://github.com/bigtreetech/PandaTouch/issues/82#issuecomment-2629741076)
> 
> I saw that the argument was closed and [@dilixi](https://github.com/dilixi) kindly posted this tool : https://github.com/bigtreetech-beta/PandaTouch_Addtion_IMG are there any instructions available? thanks and cheers all

@Winky1st 
Hi,
Please have a try with this link https://generate-addtion-img.vercel.app/
And I will add instructions for that tool.

Winky1st, 2025-03-24:
Hi @dilixi and thankyou, as usual you super kind!
I've tried your link https://generate-addtion-img.vercel.app/  and it works but only for addition. i guess is still based on old fw 1.0..6.x

I have updated my translation into italian from the newest eng version I found (instructions for drying at the eof) but fat wont work.
I enclose here below the updated italian yml file and the 2 .img (fat isn't working so you can verify it) addition is working perfectly.

when all is ready ill post all into https://github.com/bigtreetech/PandaTouch/issues/82#issuecomment-2629741076

cheers bro and enjoy!

dilixi, 2025-03-25:
> [pandatouch_Italian-V1.0.7.zip](https://github.com/user-attachments/files/19432633/pandatouch_Italian-V1.0.7.zip)
> 
> Hi [@dilixi](https://github.com/dilixi) and thankyou, as usual you super kind! I've tried your link https://generate-addtion-img.vercel.app/ and it works but only for addition. I guess it is still based on old fw 1.0.6.x
> 
> I have updated my translation into Italian from the newest English version I found (instructions for drying at the eof) but fat wont work. I attach the updated Italian yml file and the 2 .img into a zip (fat isn't working so you can verify it) addition is working perfectly.
> 
> When all will be ready I'll post results into [#82 (comment)](https://github.com/bigtreetech/PandaTouch/issues/82#issuecomment-2629741076) or where you will direct me
> 
> cheers bro and enjoy!

Hi,
It should can work now.
I've forgot to update files.

Please delete cache from browser first.
![Image](https://github.com/user-attachments/a

[pandatouch_Italian-V1.0.7_new.zip](https://github.com/user-attachments/files/19441290/pandatouch_Italian-V1.0.7_new.zip)

ssets/b5506b13-6d33-475d-b09a-b21b2b800e3d)

dilixi, 2025-03-25:
> > [pandatouch_Italian-V1.0.7.zip](https://github.com/user-attachments/files/19432633/pandatouch_Italian-V1.0.7.zip)
> > Hi [@dilixi](https://github.com/dilixi) and thankyou, as usual you super kind! I've tried your link https://generate-addtion-img.vercel.app/ and it works but only for addition. I guess it is still based on old fw 1.0.6.x
> > I have updated my translation into Italian from the newest English version I found (instructions for drying at the eof) but fat wont work. I attach the updated Italian yml file and the 2 .img into a zip (fat isn't working so you can verify it) addition is working perfectly.
> > When all will be ready I'll post results into [#82 (comment)](https://github.com/bigtreetech/PandaTouch/issues/82#issuecomment-2629741076) or where you will direct me
> > cheers bro and enjoy!
> 
> Hi, It should can work now. I've forgot to update files.
> 
> Please delete cache from browser first.
> 
> [image]
> 
> [pandatouch_Italian-V1.0.7_new.zip](https://github.com/user-attachments/files/19441290/pandatouch_Italian-V1.0.7_new.zip)
> 
> ssets/b5506b13-6d33-475d-b09a-b21b2b800e3d)

You can use pandatouch_Italian-V1.0.7_new.zip.

Neuropsihopuppu, 2025-03-25:
I modified the default english .yml file and changed some text from lowercase to upper case and used the https://generate-addtion-img.vercel.app/ to create the .img file i also changed the panda logo to my personal one.

The image changed just fine but the text did not change at all am i doing something wrong?
Also will you allow changing the bigtreetech logo on the main/idle screen at any point i feel like having 2 logos while the printer is idleing is to much since one is already printed in while on  the screen?

Thank you in advance.

Have a great day!

dilixi, 2025-03-25:
> I modified the default english .yml file and changed some text from lowercase to upper case and used the https://generate-addtion-img.vercel.app/ to create the .img file i also changed the panda [...]

Hi,
Can you show me the setting page from your panda touch?

We need to select customized language name at the setting page.

The logo cannot be changed in current firmware but we can do it in a beta version.

Neuropsihopuppu, 2025-03-25:
Oh i messed up :)) i didnt change the name,and now i have 2 english options when i select the second one the text is all broken ,can this be fixed by just reflashing the stock .img file back? /

dilixi, 2025-03-26:
> Oh i messed up :)) i didnt change the name,and now i have 2 english options when i select the second one the text is all broken ,can this be fixed by just reflashing the stock .img file back?
> 
> [image]

You can send me your yml file and I will check it later.

dilixi, 2025-03-26:
We have updated it to https://ptimgtool.bttwiki.com

dilixi, 2025-03-26:
> Oh i messed up :)) i didnt change the name,and now i have 2 english options when i select the second one the text is all broken ,can this be fixed by just reflashing the stock .img file back?
> 
> [image]

You can send me a email if you have more questions about that tool.

closed 2025-03-26 by dilixi