bigtreetech/PandaTouch #260: Panda touch crashing

JuanStecher, 2025-03-26:
Hello! My Panda Touch has a bug. When it's inactive for a long time with the screen off, when I touch it to see the printing progress, the visual interface crashes, similar to what happens when File Explorer closes and the window drags across the screen. It instantly reboots, displaying the Bigtreetech logo.
The screen has the latest firmware version 1.0.7 and its corresponding image, and this has been happening for several versions.

Rebian83, 2025-03-28:
I have exactly the same problem with the same version.

dilixi, 2025-03-31:
> WhatsApp.Video.2025-03-27.at.09.16.45.mp4 
> Hello! My Panda Touch has a bug. When it's inactive for a long time with the screen off, when I touch it to see the printing progress, the visual interface crashes, similar to what happens when File Explorer closes and the window drags across the screen. It instantly reboots, displaying the Bigtreetech logo. The screen has the latest firmware version 1.0.7 and its corresponding image, and this has been happening for several versions.

@JuanStecher @Rebian83 
Hi all, 

1 Can you reproduce it everytime?
2 Does it work with DC mode?
3 Dose it connect to printer by USB-C?
4 Do you have time to help me get some logs from your panda touch?(1284513393@qq.com)

JuanStecher, 2025-03-31:
> > WhatsApp.Video.2025-03-27.at.09.16.45.mp4
> > Hello! My Panda Touch has a bug. When it's inactive for a long time with the screen off, when I touch it to see the printing progress, the visual interface crashes, similar to what happens when File Explorer closes and the window drags across the screen. It instantly reboots, displaying the Bigtreetech logo. The screen has the latest firmware version 1.0.7 and its corresponding image, and this has been happening for several versions.
> 
> [@JuanStecher](https://github.com/JuanStecher) [@Rebian83](https://github.com/Rebian83) Hi all,
> 
> 1 Can you reproduce it everytime? 2 Does it work with DC mode? 3 Dose it connect to printer by USB-C? 4 Do you have time to help me get some logs from your panda touch?([1284513393@qq.com](mailto:1284513393@qq.com))

The error occurs every time the screen is left idle for more than 45 minutes or an hour. I currently have it running on DC 5V connected to the original USB cable that comes with the screen, with a direct connection to the P1S's display board.
Of course, I could send you the logs! If you tell me the instructions, I can send them to that email address.

JuanStecher, 2025-03-31:
> > > > WhatsApp.Video.2025-03-27.at.09.16.45.mp4
> > > > Hello! My Panda Touch has a bug. When it's inactive for a long time with the screen off, when I touch it to see the printing progress, the visual interface crashes, similar to what happens when File Explorer closes and the window drags across the screen. It instantly reboots, displaying the Bigtreetech logo. The screen has the latest firmware version 1.0.7 and its corresponding image, and this has been happening for several versions.
> > > 
> > > 
> > > [@JuanStecher](https://github.com/JuanStecher) [@Rebian83](https://github.com/Rebian83) Hi all,
> > > 1 Can you reproduce it everytime? 2 Does it work with DC mode? 3 Dose it connect to printer by USB-C? 4 Do you have time to help me get some logs from your panda touch?([1284513393@qq.com](mailto:1284513393@qq.com))
> > 
> > 
> > The error occurs every time the screen is left idle for more than 45 minutes or an hour while a print job is in progress. I currently have it running on DC 5V connected to the screen's original USB cable, with a direct connection to the P1S's display board. Of course, I could send you the logs! If you tell me the instructions, I can send them to that email address.
> 
> Hi, I will upload the instructions as soon as possible. If you change the sleep time to 2 minutes then does it can be reproduced again? If you change the power-mode to battery then does it can be reproduced again?

It's currently set to shut off in 2 minutes, but when you take it out of standby mode before 45 minutes have passed, it works normally. However, when it's been in standby mode for a long time, it does. I'm going to try battery mode, although I don't think it will change much since I'd be attached to the charging base anyway, and since the battery doesn't last more than 30 minutes, the error wouldn't occur.

dilixi, 2025-03-31:
> > > > > WhatsApp.Video.2025-03-27.at.09.16.45.mp4
> > > > > Hello! My Panda Touch has a bug. When it's inactive for a long time with the screen off, when I touch it to see the printing progress, the visual interface crashes, similar to what happens when File Explorer closes and the window drags across the screen. It instantly reboots, displaying the Bigtreetech logo. The screen has the latest firmware version 1.0.7 and its corresponding image, and this has been happening for several versions.
> > > > 
> > > > 
> > > > [@JuanStecher](https://github.com/JuanStecher) [@Rebian83](https://github.com/Rebian83) Hi all,
> > > > 1 Can you reproduce it everytime? 2 Does it work with DC mode? 3 Dose it connect to printer by USB-C? 4 Do you have time to help me get some logs from your panda touch?([1284513393@qq.com](mailto:1284513393@qq.com))
> > > 
> > > 
> > > The error occurs every time the screen is left idle for more than 45 minutes or an hour while a print job is in progress. I currently have it running on DC 5V connected to the screen's original USB cable, with a direct connection to the P1S's display board. Of course, I could send you the logs! If you tell me the instructions, I can send them to that email address.
> > 
> > 
> > Hi, I will upload the instructions as soon as possible. If you change the sleep time to 2 minutes then does it can be reproduced again? If you change the power-mode to battery then does it can be reproduced again?
> 
> It is currently set to turn off in 2 minutes, but when you take it out of sleep mode without having passed about 30 minutes it works normally, but when it spends a lot of time in sleep mode it does reproduce it. I'm going to try battery mode, although I don't think it will change much since it would be stuck to the charging base anyway, and since the battery doesn't last more than 30 minutes the error would not occur. In fact, I usually remember having left the battery mode on without realizing it a long time ago, and when waking up the screen it did not present this error, it seems to be something related to the DC 5v mode.

Thanks for your quickly response.

There is a link you can find the .bin and instructions for getting logs from panda touch.

https://github.com/bigtreetech-beta/PandaTouch-Support/tree/main/Get%20logs

JuanStecher, 2025-03-31:
> Thanks for your quickly response.
>
> There is a link you can find the .bin and instructions for getting logs from panda touch. [...]

I have already managed to connect the device to the program, and it correctly sends the logs to the system. I will perform a test print and send the logs that result in the error. Thank you.

dilixi, 2025-03-31:
> I have already managed to connect the device to the program, and it correctly sends the logs to the system. I will perform a test print and send the logs that result in the error. Thank you.

Thanks a million.

JuanStecher, 2025-03-31:
> Thanks a million.

When the screen is suspended for a long period of time and connected to the program, when trying to wake it up, xtools instantly displays the legend > Access Denied (WITHOUT THE SYSTEM CRASH), which does not happen when the screen is on and you force the restart from the UI, where it does show logs of the restart. Perhaps this makes us understand that when it is suspended for a long period of time and trying to wake it up, there is some data transmission via USB that causes this error along with the printer screen board. I have very little programming knowledge to help in anything, but I can give my contribution by commenting on my experience with the screen.

bigtreetech, 2025-03-31:
@dilixi I wonder whether it would make a difference to gradually fade the screen brightness up over a 2s period after waking. It could be that the sudden wake is causing too much current drain from the P1S port?

JuanStecher, 2025-04-01:
@bigtreetech @dilixi here we have an electrical problem, I was trying a couple of things with the things that were mentioned, I have left the printer at rest with the led light on, my p1s has the biqu panda lux mod, which respects the same amount of consumption as the original led, and this original led was removed to not cause excessive consumption on the board, the thing is that the screen did restart when I wanted to wake it up with the light on after several hours, and I repeated the process but turning off the light from bambu handy so as not to wake up the screen, and when waking it up it did not restart, perhaps it is a sudden excess of consumption, which added to the led lamp on overloaded the board for an instant.

dilixi, 2025-04-07:
> [@bigtreetech](https://github.com/bigtreetech) [@dilixi](https://github.com/dilixi) here we have an electrical problem, I was trying a couple of things with the things that were mentioned, I have [...]

It should be an electrical problem.
Can you reproduce this problem in your way?

JuanStecher, 2025-04-25:
I have updates, right now it's restarting every 30-45 seconds automatically all the time without stopping, I'm considering uninstalling it and using the original screen, I think my unit is faulty :(

dilixi, 2025-04-25:
> I have updates, right now it's restarting every 30-45 seconds automatically all the time without stopping, I'm considering uninstalling it and using the original screen, I think my unit is faulty :(

You can try this one:
https://github.com/bigtreetech/PandaTouch/issues/281

I think there are two issues and they are different.

bigtreetech, 2025-04-25:
> I have updates, right now it's restarting every 30-45 seconds automatically all the time without stopping, I'm considering uninstalling it and using the original screen, I think my unit is faulty :(

This is due to a recent change in the Bambu cloud and how it handles thumbnails. There is a new beta to resolve it.

davidrenji, 2025-04-26:
Hi All, I have FW 1.0.7.1 and I'm having the original issue mentioned in this thread, the screen crashes after a long period of being idled while printing. 
Does this beta (v1.0.7.2_beta2.bin) has a fix for that?

bigtreetech, 2025-04-26:
> Hi All, I have FW 1.0.7.1 and I'm having the original issue mentioned in this thread, the screen crashes after a long period of being idled while printing. Does this beta (v1.0.7.2_beta2.bin) has a fix for that?

Yes. Try it.

JuanStecher, 2025-04-28:
> > Hi All, I have FW 1.0.7.1 and I'm having the original issue mentioned in this thread, the screen crashes after a long period of being idled while printing. Does this beta (v1.0.7.2_beta2.bin) has a fix for that?
> 
> Yes. Try it.

Where can I find this beta to download and try?

JuanStecher, 2025-04-28:
> > Hi All, I have FW 1.0.7.1 and I'm having the original issue mentioned in this thread, the screen crashes after a long period of being idled while printing. Does this beta (v1.0.7.2_beta2.bin) has a fix for that?
> 
> Yes. Try it.

https://github.com/user-attachments/assets/dfeb7844-68df-4ad3-a2ca-c2bdeed0b5ec

I'm testing the beta, although it no longer restarts itself every 30 seconds, the same thing happens when it's suspended.

bigtreetech, 2025-04-28:
> > > Hi All, I have FW 1.0.7.1 and I'm having the original issue mentioned in this thread, the screen crashes after a long period of being idled while printing. Does this beta (v1.0.7.2_beta2.bin) has a fix for that?
> > 
> > 
> > Yes. Try it.
> 
>  Video.de.WhatsApp.2025-04-28.a.las.12.23.19_3469f5c4.mp4 
> I'm testing the beta, although it no longer restarts itself every 30 seconds, the same thing happens when it's suspended.

Please help to clarify the points below:

1. This is still only happening after the screen has been off for greater than 30 minutes but less than that will not cause this to happen?
2. You have it plugged into the PSU of the printer (USB port) using the original cable and it is running in DC mode?

JuanStecher, 2025-04-28:
> Please help to clarify the points below:
>
> 1. This is still only happening after the screen has been off for greater than 30 minutes but less than that will not cause this to happen?
> 2. You have it [...]

In fact, this only happens after leaving the screen suspended for approximately 30 minutes. Once it restarts, if only a few minutes pass after it's suspended, it will wake up correctly. The screen is connected to the correct printer port, as indicated in the manual, in DC mode.

dilixi, 2025-04-29:
> In fact, this only happens after leaving the screen suspended for approximately 30 minutes. Once it restarts, if only a few minutes pass after it's suspended, it will wake up correctly. The screen [...]

Hi,
If you can reproduce that problem 100% we can do a test with a special firmware.
Does it reboot every time when it sleeped for more than 30 minutes?

JuanStecher, 2025-04-29:
I can confirm that it is 100% of the time that it remains suspended, so I can try any version of test firmware to solve this error.

> Hi, If you can reproduce that problem 100% we can do a test with a special firmware. Does it reboot every time when it sleeped for more than 30 minutes?

I can confirm that it is 100% of the time that it remains suspended for more than 30 minutes, so I can try any test firmware version to help resolve this error.

JuanStecher, 2025-04-29:
https://github.com/user-attachments/assets/6fc1c392-bf98-4ac5-b6e5-1771d50938f7

This just happened to me with the new beta 1.0.7.2,

JuanStecher, 2025-04-29:
https://github.com/user-attachments/assets/b98a153d-6d21-4742-a7f1-9d390655b7fc

This just happened to me with the new beta 1.0.7.2

dilixi, 2025-04-29:
> I have already managed to connect the device to the program, and it correctly sends the logs to the system. I will perform a test print and send the logs that result in the error. Thank you.

Hi,
Did you send me the logs before?

I have send you a special firmware before.
but I dont receive any logs about that.

I will send you a new firmware.

JuanStecher, 2025-04-29:
> Hi, Did you send me the logs before?
>
> I have send you a special firmware before. but I dont receive any logs about that.
>
> I will send you a new firmware.

I have not been able to send logs because when connecting it via USB to a computer the error is not translated, above in the thread we agreed that the problem occurs with the USB connection of the P1S, as if after a while the port was disabled and when requesting a power surge to reactivate the screen it causes it to crash, I do not have much knowledge in electronics and what I am saying may have nothing to do with it, but I am open to trying different firmwares in order to help solve this error.

bigtreetech, 2025-04-29:
> I have not been able to send logs because when connecting it via USB to a computer the error is not translated, above in the thread we agreed that the problem occurs with the USB connection of the [...]

Ah, then it could be a problem with the USB power supply on your P1S. It may be cutting the current off for some reason. There is a resettable fuse within their system. It could be that the fuse is momentarily tripping on your machine. If this is the case then you can either try a different, fatter cable or you may need to run it from an external supply.

dilixi, 2025-04-29:
> > > Hi All, I have FW 1.0.7.1 and I'm having the original issue mentioned in this thread, the screen crashes after a long period of being idled while printing. Does this beta (v1.0.7.2_beta2.bin) has a fix for that?
> > 
> > 
> > Yes. Try it.
> 
> I upgraded to v1.0.7.2_beta2 but as mentioned by other users, the issue of the screen rebooting after 30min of being on standby mode while printing, is still occurring. I have the screen plugged to the P1S USB port as well.

Beta2 didn't fix that issues.
It only fixed the bug caused by cloud mode.

So, there are two issues, we just resolve one of them.

This issue should related to hardware.

Services8810, 2025-04-29:
Hi,

I have exactly the same behavior as the 1st post. The screen restarts after if I leave it idle for long period. This does not create issue with the prints as the screen returns correctly to the printing screen but I saw today that while I have left the printer drying filament, the temp was correct but the drying time was not measuring anymore and the countdown has stopped. I am at 1.0.7.1 but I had this issue with the 1.0.6 version also. I have the screen connected with the usb supplied and at DC.

Any help?

JuanStecher, 2025-04-29:
> > Video.de.WhatsApp.2025-04-29.a.las.08.44.04_c1f5224c.mp4
> > This just happened to me with the new beta 1.0.7.2, after a few minutes it was restored
> 
> From this video. Did you switch the power mode to battery?

I just tested it in battery mode, connected to the charging base mounted on the printer, and the error does NOT occur. The error is exclusive to DC mode. I hope there's a possible solution, as it's a shame to constantly cycle the battery when using the printer, and then cause worse damage when the screen turns off when the battery dies with the printer off.

dilixi, 2025-04-30:
> I just tested it in battery mode, connected to the charging base mounted on the printer, and the error does NOT occur. The error is exclusive to DC mode. I hope there's a possible solution, as it's [...]

V1.0.6 doesn't have drying feature.

What is the sleep time of the original screen?
If you can reproduce that problem 100% and please connect the Panda Touch to a PC via USB-C.
It can reduce the current of whole circult.

I think the Panda Touch will not reboots.

Services8810, 2025-05-01:
> V1.0.6 doesn't have drying feature.
> 

When I say I had the same issue with 1.0.6, I am reffering to the crash and not the drying feature

dilixi, 2025-05-01:
> > V1.0.6 doesn't have drying feature.
> 
> When I say I had the same issue with 1.0.6, I am reffering to the crash and not the drying feature

I think you can use another power supply to do a test.
You can  connect the Panda Touch to a phone charger via usb and do the same thing.

It relates to the current.

There is a special firmware which may reduce some current when Panda Touch wake up.

But it should not be a bug from Panda Touch.

https://github.com/bigtreetech-beta/PandaTouch-Support/blob/main/firmware/v1.0.7.3_log.bin

davidrenji, 2025-05-01:
Just to add a little bit of more info, I have the "LED Upgrade Kit Illuminate the Print Area with Simple Mounting for Bambu Lab P1/X1/A1 Series". And I always have it on while printing. But yesterday I did a 3 hours print and I turn it off from the start, and when the printing was finished, I woke up the panda screen, and it didn't crash. So I think that give us more clues to know that the issue is related to the power consumption as it was mentioned before.
Later I will try FW v.1.0.7.3, thanks.

AfterburnNL, 2025-05-04:
> I have the biqu panda lux, which completely replaces the original light, consuming the same amount of energy. It happens to me whether I leave that LED on or off. I don't think it's a problem with the LED's consumption, but rather with the panda touch. I will also try firmware 1.0.7.3.

Did 1.0.7.3 work for you? 

Don't have the Panda Lux, but I have been seeing the same issue since upgrading the firmware to 1.0.7.1 and higher. The 1.0.6.3 firmware didn't have this bug. So when reverting back to that version the problem disappeared.

Update: Just now went forward again to 1.0.7.2 and the Panda Touch crashes every single time it is woken from sleep.

JuanStecher, 2025-05-04:
> > > I have the biqu panda lux, which completely replaces the original light, consuming the same amount of energy. It happens to me whether I leave that LED on or off. I don't think it's a problem with the LED's consumption, but rather with the panda touch. I will also try firmware 1.0.7.3.
> > 
> > 
> > Did 1.0.7.3 work for you?
> > Don't have the Panda Lux, but I have been seeing the same issue since upgrading the firmware to 1.0.7.1 and higher. The 1.0.6.3 firmware didn't have this bug. So when reverting back to that version the problem disappeared.
> > Update: Just now went forward again to 1.0.7.2 and the Panda Touch crashes every single time it is woken from sleep.
> 
> Did you try 1.7.3?

I have tried with firmware 1.0.7.3, the result is the same, it restarts again, it seems that they have added the option that when touching the brightness it slowly increases, but anyway it worked about 3 times and then it started to restart as always.

AfterburnNL, 2025-05-04:
> Did you try 1.7.3?

Gave 1.0.7.3 a shot and it solves the problem. Good solution to gradually increase the screen brightness to draw less power on wake.
When can we expect an official release?

bigtreetech, 2025-05-04:
> > > > I have the biqu panda lux, which completely replaces the original light, consuming the same amount of energy. It happens to me whether I leave that LED on or off. I don't think it's a problem with the LED's consumption, but rather with the panda touch. I will also try firmware 1.0.7.3.
> > > 
> > > 
> > > Did 1.0.7.3 work for you?
> > > Don't have the Panda Lux, but I have been seeing the same issue since upgrading the firmware to 1.0.7.1 and higher. The 1.0.6.3 firmware didn't have this bug. So when reverting back to that version the problem disappeared.
> > > Update: Just now went forward again to 1.0.7.2 and the Panda Touch crashes every single time it is woken from sleep.
> > 
> > 
> > Did you try 1.7.3?
> 
> Gave 1.0.7.3 a shot and it solves the problem. Good solution to gradually increase the screen brightness to draw less power on wake.
> 
> Tried it a couple of times the last hour and so far no crashes.

It seems as if the max power output on the USB port of the P1S may degrade with time. There is resettable fuse on the output and they tend to increase in resistance after having been reset a few times so this may be the cause. We are working on other ways to reduce the wake up draw in 1.7.3 to prevent the fuse from resetting and we are also adding a few more features. Final release will probably be in 2-3 weeks.

JuanStecher, 2025-05-04:
> It seems as if the max power output on the USB port of the P1S may degrade with time. There is resettable fuse on the output and they tend to increase in resistance after having been reset a few [...]

I understand, I'm glad a solution has been found, I'll be happy to test the final version when it's developed to contribute to the community's resolution of this (this is the grain of sand I can contribute) and thank you very much for giving importance to this issue.

davidrenji, 2025-05-05:
> I understand, I'm glad a solution has been found, I'll be happy to test the final version when it's developed to contribute to the community's resolution of this (this is the grain of sand I can [...]

Same here, if you have a version that we can test, I gladly can help testing. Thanks.

bigtreetech, 2025-05-05:
> Same here, if you have a version that we can test, I gladly can help testing. Thanks.

You tried 1.7.3 already?

davidrenji, 2025-05-05:
> > Same here, if you have a version that we can test, I gladly can help testing. Thanks.
> 
> You tried 1.7.3 already?

Yes, I've tried 1.7.3 this weekend and I did 2 prints that were like 40min each, I did 2 that were like 2h each.
And every time the print finished, I woke up the screen and it didn't restarted.

bigtreetech, 2025-05-05:
> Yes, I've tried 1.7.3 this weekend and I did 2 prints that were like 40min each, I did 2 that were like 2h each. And every time the print finished, I woke up the screen and it didn't restarted.

OK so it looks like it is working for you too then. Great stuff.

dilixi, 2025-05-06:
@JuanStecher 
Hi,
You said you have tried with v1.0.7.3_log.bin but that doesn't work for you.
Please config the backlight to 10% and try again.
thanks.

davidrenji, 2025-05-06:
Update: Today I was doing a 2h printing, and after 1h while printing I woke up the screen and it crashed again.
I will decrease the backlight to 50% and do more testing.

davidrenji, 2025-05-07:
Another update: I tried 50% backlight and still got the crash, then I tried 10% and still got the crash. What is weird is that now, the crash happened every time even waking the screen in prints that lasted 15 min.

dilixi, 2025-05-07:
> Another update: I tried 50% backlight and still got the crash, then I tried 10% and still got the crash. What is weird is that now, the crash happened every time even waking the screen in prints that lasted 15 min.

Hi,
Please have a try with this one.
https://github.com/bigtreetech-beta/PandaTouch-Support/blob/main/firmware/v1.0.7.2_bl1.bin

When I removed the original screen during the Panda Touch cycle restart, the Panda Touch did not restart again. 
It should be the current was too high and the voltage was lowered after connecting the Panda Touch.

davidrenji, 2025-05-07:
> Hi, Please have a try with this one. https://github.com/bigtreetech-beta/PandaTouch-Support/blob/main/firmware/v1.0.7.2_bl1.bin
>
> When I removed the original screen during the Panda Touch cycle [...]

I'm testing v1.0.7.2_bl1.bin, but with this FW, the screen never goes to sleep. (I have it set to go sleep after 2 min)

dilixi, 2025-05-07:
> I'm testing v1.0.7.2_bl1.bin, but with this FW, the screen never goes to sleep. (I have it set to go sleep after 2 min)

Yes.
The backlight will be changed to 1% when it is going to sleep.

You can wake up it after 30 minutes.

davidrenji, 2025-05-07:
> Yes. The backlight will be changed to 1% when it is going to sleep.
>
> You can wake up it after 30 minutes.

Mmm I've been testing it the entire morning and the screen backlight never changes.

dilixi, 2025-05-07:
> Mmm I've been testing it the entire morning and the screen backlight never changes.

Okay.
Thank you.

dilixi, 2025-05-08:
> Mmm I've been testing it the entire morning and the screen backlight never changes.

https://github.com/user-attachments/assets/de3270cc-fb5c-4265-b9ef-5db8a30cb1d3

Hi,
I have tested it again.

You can see that the brightness is getting higher and higher when I woke up it.  

You can set the backlight to 80% before you try again.

davidrenji, 2025-05-13:
> Video.de.WhatsApp.2025-04-28.a.las.12.23.19_3469f5c4.mp4
>
> I'm testing the beta, although it no longer restarts itself every 30 seconds, the same thing happens when it's suspended.

To be honest, in my case, I don't see a change in the backlight brightness. It looks like it's always on

dilixi, 2025-05-14:
> To be honest, in my case, I don't see a change in the backlight brightness. It looks like it's always on

Can you show me the setting page of Panda Touch?
Set change the backlight to 80%.
Set the sleep time to 2min.

dilixi, 2025-05-25:
> The issue is there, the 1.0.7.3 file not 😂

Hi,
There is a link which you can get 1.0.7.3.
https://github.com/bigtreetech-beta/PandaTouch-Support/blob/aff6a66430b9eec9db08586e89a86fbd28b3096f/firmware/v1.0.7.3_log.bin

Services8810, 2025-08-02:
I updated my Panda Touch at v1.0.7.3. Unfortunately the screen still has the same behavior and crash after idle for some minutes.

Any news on that?

Services8810, 2025-09-04:
> The panda touch crashing looks like it's caused by the screen requiring more power than the P1 USB port can deliver. Today I connected the touch to a standard USB charger and it works fine. Although I hate throwing more money at BTT for the poor power design I'll probably get the branch usb supply to power this thinng.

I did exactly the same (external power supply) and solve the issue

rocket59, 2025-09-29:
I installed my new Panda branch a few days ago and the touch has been rock solid ever since. This is not a software bug just poor hardware design.

Services8810, 2025-10-01:
> Is your power supply at least 2A?

I checked today, it is 3A