bigtreetech/PandaTouch #261: pandatouch.us.kg is not reachable

mytechguyri, 2025-03-28:
Trying to access the panda image update tool as directed in the various readme files at https://pandatouch.us.kg but not accessible, DNS is not returning an A record.

:~$ dig pandatouch.us.kg

; <<>> DiG 9.18.33-1~deb12u2-Debian <<>> pandatouch.us.kg
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NXDOMAIN, id: 48702
;; flags: qr rd ra; QUERY: 1, ANSWER: 0, AUTHORITY: 1, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 1232
;; QUESTION SECTION:
;pandatouch.us.kg.              IN      A

;; AUTHORITY SECTION:
us.kg.                  3198    IN      SOA     ns1.dpdns.org. admin.dpdns.org. 2025032393 3600 1800 1209600 86400

;; Query time: 60 msec
;; SERVER: 172.22.224.1#53(172.22.224.1) (UDP)
;; WHEN: Fri Mar 28 03:27:58 EDT 2025
;; MSG SIZE  rcvd: 114

dilixi, 2025-03-31:
> Trying to access the panda image update tool as directed in the various readme files at https://pandatouch.us.kg but not accessible, DNS is not returning an A record.
>
> :~$ dig pandatouch.us.kg
>
> ; [...]

Hi,
Please use this link.

https://ptimgtool.bttwiki.com/

mytechguyri, 2025-04-02:
That works, thank you... documentation/wiki should be updated to reflect the correct URL

closed 2025-04-02 by mytechguyri