bigtreetech/PandaTouch #262: Filament drying option not working

mytechguyri, 2025-03-28:
I am running printer firmware 1,07 on my P1S
i am running panda touch firmware V1.0.7_Mar 22 2025_09:40:32
When I select drying on the filament screen, I get a message 'not support on A1 a1 mini and p1p"
My printer is a p1s not an A1, A1 mini or P1P
i've attempted re-installing the Panda Touch firmware and the en.img file
i've attempted downgrading my p1s to 1.0.6.2 firmware
still filament drying gives me the "not support A1, A1-Mini and p1p" message, and does not allow filament drying to function.

Neuropsihopuppu, 2025-03-28:
delete and re-add the printer that worked for me

mytechguyri, 2025-03-29:
> delete and re-add the printer that worked for me

Tried that too... no luck

macman-3D, 2025-03-29:
Same here, deleted and re-added the P1S from the Panda Display and not working for me either. I have a second P1S with Panda Touch with same firmware on the P1S and that one is working fine. I leaning on doing a factory reset on the Panda Touch, but before I do that, I want to hear other options if available.

dilixi, 2025-03-31:
> > delete and re-add the printer that worked for me
> 
> Tried that too... no luck

@mytechguyri @macman-3D 
Hi all,
Can you send me a shotscreen which contains the detail information of printer from panda touch?

[image]

macman-3D, 2025-03-31:
Apparently the Printer wasn't connected to the cloud so the model (P1S) did not show. Once connected, P1S was displayed and was able to use the Filament drying option. Thanks again for the screenshot!

mytechguyri, 2025-04-02:
Printer WAS connected to the cloud, and model DID show P1S and it STILL did not work.   
I did manager to discover the problem however... I believe this is in fact a bug in the new firmware

If I add the printer by selecting SCAN and allowing it to find the printer on the network on its own, it works 
If however I add the printer MANUALLY by entering the IP address, model (P1S) and password and serial number all manually, it will happilly add and connect to the printer, but filament drying option does not work.

dilixi, 2025-04-05:
> Printer WAS connected to the cloud, and model DID show P1S and it STILL did not work. I did manager to discover the problem however... I believe this is in fact a bug in the new firmware
>
> If I add [...]

Hi,
Can you upload a video or some screenshot to show me the  status of connection and the information of printer?

raihei, 2025-04-06:
Same for me, I had to delete the printer and to use "scan" to add it back. It was not connected to cloud before and is now also not connected to cloud. I have no idea, if I added it in the past manually or by scanning. But now it works.

jakesza, 2025-04-08:
I have several P1S’s, only 1 does this which does not have an AMS. The model displays correctly but getting the error not compatible.

dilixi, 2025-04-08:
> I have several P1S’s, only 1 does this which does not have an AMS. The model displays correctly but getting the error not compatible.

@jakesza 
Hi,
Can you upload a picture like this?
We need to confirm 2 points
1 model picture
2 model name

[image]

jakesza, 2025-04-08:
> > I have several P1S’s, only 1 does this which does not have an AMS. The model displays correctly but getting the error not compatible.
> 
> [@jakesza](https://github.com/jakesza) Hi, Can you upload a picture like this? We need to confirm 2 points 1 model picture 2 model name
> 
> [image]

Thank you. I set the printer manually to P1S and it works.

dilixi, 2025-04-08:
> Printer WAS connected to the cloud, and model DID show P1S and it STILL did not work. I did manager to discover the problem however... I believe this is in fact a bug in the new firmware
>
> If I add [...]

@mytechguyri 

Hi,
If you select the model by click the icon then it should be correcty.
There is a firmware you can have a try with it thanks.

https://github.com/bigtreetech-beta/PandaTouch-Support/blob/main/firmware/v1.0.7_fix_dry_filament.bin