bigtreetech/PandaTouch #263: Filament Drying on P1P

bl4ckm0onk3y, 2025-03-28:
Hello, i have P1P upgraded fully to P1S and more upgrades than stock. 
The new filament drying feature isn't available despite the upgrades. It's not a big issue since i have dedicated hardware for drying, but if someone  is in the same situation like me and needs to use this feature he can't

dilixi, 2025-03-31:
> Hello, i have P1P upgraded fully to P1S and more upgrades than stock. The new filament drying feature isn't available despite the upgrades. It's not a big issue since i have dedicated hardware for drying, but if someone is in the same situation like me and needs to use this feature he can't

Hi,
Could you please change the Printer Type from the setting page of the printer?
Just change it to P1S and have a try again.

https://github.com/user-attachments/assets/0649417e-e612-4c23-99fb-76fe5a79c1ed

bl4ckm0onk3y, 2025-04-04:
Changing the printer model in the setting by edit feature unlocked the filament drying.