bigtreetech/PandaTouch #265: Panda Touch v1.1.1

porkydja, 2025-04-07:
Does anyone know what is the difference between the first initial v1.0 and the v1.1.1?

dilixi, 2025-04-10:
> Does anyone know what is the difference between the first initial v1.0 and the v1.1.1?

Hi,
The latest version so far is 1.0.7.

Please check this link. 
https://bttwiki.com/PandaTouch.html#firmware-history

bigtreetech, 2025-04-10:
> My question is not about the firmware. I have the Panda Touch v1.0 that already runs the latest firmware. My question was about the device version, what is the difference in a hardware between v1.0 and v1.1.1( that is being sold now)

There have been minor changes to the hardware for the sake of streamlining production within the factory. Nothing that impacts the end user though.

closed 2025-04-10 by bigtreetech

porkydja, 2025-04-11:
Thanks for the answer! It’s clear now