bigtreetech/PandaTouch #274: Bug Report: Editing the target temperature is invalid.

CatCw, 2025-04-18:
When the printer is cooling down after completing the print, editing the target temperature is ineffective (the displayed value can be changed, but the printer will not heat up), but it is possible to do so on the original screen or through the APP.

dilixi, 2025-04-18:
> When the printer is cooling down after completing the print, editing the target temperature is ineffective (the displayed value can be changed, but the printer will not heat up), but it is possible to do so on the original screen or through the APP.

Did you login an account on Panda touch when you try to editing the target temperature?
Do you mean the bug can be reproduced only after completing the print?

CatCw, 2025-04-20:
1. Do you mean the account of BambuLab？If so, the answer is No.
2. I did more tests. It doesn't work in other situations now either. (For both Printer and PandaTouch, I'm using the latest release firmware.)

dilixi, 2025-04-20:
> 1. Do you mean the account of BambuLab？If so, the answer is No.
> 2. I did more tests. It doesn't work in other situations now either. (For both Printer and PandaTouch, I'm using the latest release firmware.)

Hi,
bambulab have changed something.
1 LAN Only Off
we need control printer to heat with an account

2 LAN Only On
we can control printer to heat without an account

For detail:
https://bttwiki.com/PandaTouch.html#important-note

closed 2025-04-20 by dilixi