bigtreetech/PandaTouch #277: сделайте русский язык  пожалуйста

Enclave10, 2025-04-22:


dilixi, 2025-04-22:
[ru_not_finished.zip](https://github.com/user-attachments/files/19844797/ru_not_finished.zip)

Hi,
You can use this one which I get from [PR](https://github.com/bigtreetech/PandaTouch/pull/250)

You can edit it to by yourself and generated a img file by this [tool](https://ptimgtool.bttwiki.com/)

dilixi, 2025-04-22:
> No

[ru_not_finished.zip](https://github.com/user-attachments/files/19844797/ru_not_finished.zip)

Hi,
You can use this one which I get from https://github.com/bigtreetech/PandaTouch/pull/250

You can edit it by yourself and you can also  generate a img file with this [tool](https://ptimgtool.bttwiki.com/)

Enclave10, 2025-04-22:
Спасибо

closed 2025-09-29 by xperiments