bigtreetech/PandaTouch #281: Device restarts every few seconds

Neuropsihopuppu, 2025-04-24:
I updated the panda touch to 1.0.7.1 today and device keeps reseting every about 10 seconds or so

dilixi, 2025-04-24:
> I updated the panda touch to 1.0.7.1 today and device keeps reseting every about 10 seconds or so

Hi,
Please flash back to 1.0.6.3 and have a try again.

Neuropsihopuppu, 2025-04-24:
how exactly do i do that?

Neuropsihopuppu, 2025-04-24:
using the recovery tool or just the normal .bin file?

dilixi, 2025-04-24:
> how exactly do i do that?

You can use these files.
https://github.com/bigtreetech/PandaTouch/tree/master/Firmware/1.0.6.3

https://github.com/bigtreetech/PandaTouch/blob/master/Firmware/1.0.6.3/fat_panda_touch-v1.0.6.img

https://github.com/bigtreetech/PandaTouch/blob/master/Firmware/1.0.6.3/panda_touch-v1.0.6.3.bin

Neuropsihopuppu, 2025-04-24:
ok so you fould like me to flash firmware 1.0.6.3 and tyhen 1.0.7.1 after is that correct?

dilixi, 2025-04-24:
Hi,
You can have a try with disabled cloud mode first.

Have a nice day,
Jonny Chan

dilixi, 2025-04-24:
> ok so you fould like me to flash firmware 1.0.6.3 and tyhen 1.0.7.1 after is that correct?

Just do a test with 1.0.6.3 to see if there is a same problem.

Neuropsihopuppu, 2025-04-24:
turning off cloud mode does fix the problem but its not an ideal solution for me,is there any other sollution?

dilixi, 2025-04-24:
> turning off cloud mode does fix the problem but its not an ideal solution for me,is there any other sollution?

Yes.
I will solve this problem.

May I use your account to do a test tomorrow?

Neuropsihopuppu, 2025-04-24:
yes! that is no problem

dilixi, 2025-04-24:
> yes! that is no problem

what time is suitable for me to send you a message via email tomorrow?

I need you a verify code and your email address.

You can send me a email now(1284513393@qq.com)

superceleron, 2025-04-24:
my device started doing the same yesterday, it stops if i do a factory reset and then dont use cloud mode... if i enable it again it starts again to reboot.

dilixi, 2025-04-24:
> my device started doing the same yesterday, it stops if i do a factory reset and then dont use cloud mode... if i enable it again it starts again to reboot.

There should be something changed on cloud.
I will check it out tomorrow.

dilixi, 2025-04-25:
@Neuropsihopuppu 
@superceleron 

Hi all,
The format of thumbnail has been changed to jpeg recently.

Please have a try with this one:
https://github.com/bigtreetech-beta/PandaTouch-Support/blob/main/firmware/v1.0.7.2_beta2.bin

superceleron, 2025-04-25:
@dilixi its fixed for me.

dilixi, 2025-04-25:
> Same here, updated to 1.0.7.1 and it restarts every few seconds. Flashed back to 1.0.6.3, same issue. Flashed to 1.0.5.1 and its fine, flashed from 1.0.5.1 to 1.0.7.1 and same, reboots.
> 
> Staying on 1.0.5.1 for now until a fix is rolled or the cause identified.

Yes.
1.0.5.1 is fine.

You can have a try with this one:
https://github.com/bigtreetech-beta/PandaTouch-Support/blob/main/firmware/v1.0.7.2_beta2.bin

macman-3D, 2025-04-25:
I updated the panda touch to 1.0.7.1 yesterday and had the same with one of panda touch displays.  Flashed back to 1.0.6.3, same issue. Updated to v1.0.7.2 beta 2 and that fixed the reboot issue for me as well.

TaylorNQT, 2025-04-25:
> my device started doing the same yesterday, it stops if i do a factory reset and then dont use cloud mode... if i enable it again it starts again to reboot.

This is the exact same issue I am having. I was on 1.6.x, have updated to 1.7.1, and it only restarts after I connect to cloud mode.

TaylorN15, 2025-04-25:
> my device started doing the same yesterday, it stops if i do a factory reset and then dont use cloud mode... if i enable it again it starts again to reboot.

This is the exact same issue I am having. I was on 1.6.x, have updated to 1.7.1, and it only restarts after I connect to cloud mode.

noizzze00, 2025-04-28:
let't just say the trouth bambu lab sayd to this guys, look the day we make work the cloud your device stop corking, and guess what we should ask a refund from this guys

dilixi, 2025-04-28:
> let't just say the trouth bambu lab sayd to this guys, look the day we make work the cloud your device stop corking, and guess what we should ask a refund from this guys

Panda Touch might have hurt X1 series sales by giving P1S owners a cheaper way to get high-end features.

maple-underscore, 2025-05-04:
@dilixi using 1.0.7.2_beta2 works even with cloud mode, very happy that this is fixed

noizzze00, 2025-05-04:
you know if bambu wanted to do , see what kind of screen they piu on A series, at list they coud make a touch like that ad upgrade for P series users

closed 2025-05-07 by dilixi