bigtreetech/PandaTouch #282: Restart upon opening file menu

maple-underscore, 2025-04-25:
When I open the file tab to look at the MicroSD, the Panda Touch loading circle counts to around 18-20 and then reboots. I assume this had something to do with the files not appearing in root issue?

dilixi, 2025-04-25:
> When I open the file tab to look at the MicroSD, the Panda Touch loading circle counts to around 18-20 and then reboots. I assume this had something to do with the files not appearing in root issue?

Did you enable cloud mode on panda touch?

maple-underscore, 2025-04-25:
Yes, I did. i just updates to 1.0.7.1 also. Update: the rebooting happens
randomly and not just when loading files

dilixi, 2025-04-25:
> Yes, I did. i just updates to 1.0.7.1 also. Update: the rebooting happens
> randomly and not just when loading files
> […](#)

Please disable it and try again.
It should be a bug from cloud.

maple-underscore, 2025-04-25:
Will this be fixed with cloud mode in the future?
Also, I have noticed that the Panda Touch frequently disconnects from the network router (the router is in the same room).

dilixi, 2025-04-25:
> Will this be fixed with cloud mode in the future? Also, I have noticed that the Panda Touch frequently disconnects from the network router (the router is in the same room).

Yes.
It has been fixed.
Lets check out them one by one.

You can try this one first:
https://github.com/bigtreetech-beta/PandaTouch-Support/blob/main/firmware/v1.0.7.2_beta2.bin

https://github.com/bigtreetech/PandaTouch/issues/281

maple-underscore, 2025-04-25:
Got it!

closed 2025-05-04 by maple-underscore