bigtreetech/PandaTouch #284: Will not connect to a network

WarEagle93, 2025-04-25:
My Panda Touch will not connect to my network.  I was upgrading my firmware, uploaded the first file.  The Touch rebooted, now it wont reconnect to the network.  I'm just going to throw it  away the Touch at this point as there is no way to reset it to factory.  I'll never buy another one.

bigtreetech, 2025-04-25:
The panda touch should retain network settings but if it does not then it will launch its own access point which allows you to connect directly to it and complete the update. Look for an access point that is being hosted by it. If that fails then you can follow the recovery process documented in here. Good thing you'll never have to buy another one since recovery is simple.

[image]

WarEagle93, 2025-04-25:
Instructions in the recovery zip worked like a charm. Now it just wont connect to wifi LOL.  But that's a separate issue I guess.  Thanks!

bigtreetech, 2025-04-25:
Reboot your router. It could be that the actual network is being difficult.