bigtreetech/PandaTouch #294: Feature Request: Display always on during print

tpretz, 2025-04-27:
Hi,

I can see you have a display sleep timer setting.
What would be really useful, is if the display would remain on while any of the monitored / displayed printers are running a print job. 

Then revert to the current display sleep timer if they are not. 

Thanks

bigtreetech, 2025-04-27:
You really don't want this. Some prints last days on end. If you leave the screen on for that duration you will certainly get burn in permanently embedded into the screen.

tpretz, 2025-04-27:
I don't tend to run long prints. I have the panda touch located on a desk to monitor for any issues / progress (and have to wake it from time to time). 

But, i get the concern, especially as if such an option was provided, some users would have long duration prints. 

Apologies if this is existing behaviour / already configurable, i don't recall as tend to keep the screen active when i'm nearby but what about "wake on state change" ?
Having the screen sleep counter reset on any printer status change (printing -> done / printing -> error) would be useful to draw attention that something has happened

CatCw, 2025-04-28:
@bigtreetech , Please refer to the requirements #268  and #273 .

tpretz, 2025-04-28:
Thanks

Closing in favour of #268

closed 2025-04-28 by tpretz