bigtreetech/PandaTouch #297: P1S new firmware version 01.08.00.00 released - Safe to update?

macman-3D, 2025-04-29:
Is it safe to update P1S with new Firmware without affecting the Panda Touch?

version 01.08.00.00 update information :
# Things to know before updating
In order to use all the new features introduced by the new firmware update, Bambu Studio and Bambu Handy should be updated to the latest version available:
- Bambu Studio: v2.0.3.54 or later
- Bambu Handy: v3.1.0 or later
# New Features
1. Added support for multi-plate printing
- You can now select "Print all" from the Bambu Studio interface, or you can use the Multi-Plate printing operation from the printer's screen.
2. Added support for AMS 2 Pro and AMS HT
- The printer now supports starting the filament drying operation from the printer's screen
3. Added Portuguese language
4. Added Korean language
# Note
1. Please update the printer firmware before connecting the printer to the AMS 2
Pro/HT
2. Each P1 series printer supports up to four AMS units in total, regardless of type.
You can mix and match different AMS models, such as the AMS, AMS 2 Pro, and AMS HT, within this limit.Firm

macman-3D, 2025-04-29:
> The panda touch will only retain full functionality if operating in LAN - developer mode with this firmware. If operating in cloud mode it will lose the ability to control functions on the printer and only be able to read data.

Ok, thanks for reply - is LAN-developer mode option appear on this new firmware via P1S?

keithwkx, 2025-04-29:
> The panda touch will only retain full functionality if operating in LAN - developer mode with this firmware. If operating in cloud mode it will lose the ability to control functions on the printer and only be able to read data.

so does this meant the custom filament function introduced via cloud will also not work?

keithwkx, 2025-04-29:
> It seems as if this version is safe to use based on reports we have been getting but we are still waiting for it to become available on our machines to verify.

noted on this , do keep us up to date on this. Thanks!

macman-3D, 2025-04-29:
> > It seems as if this version is safe to use based on reports we have been getting but we are still waiting for it to become available on our machines to verify.
> 
> noted on this , do keep us up to date on this. Thanks!

Same here - currently running on V1.0.7.2_Apr 29 2025_16:58:04 PT firmware - thanks

DubStepMad, 2025-04-29:
I installed the firmware update v01.08.00.00 and not having issues at the moment apart from the reprint function not working as it gets stuck on autobed leveling.

This issue also happened on the previous versions of Panda Touch.

keithwkx, 2025-04-29:
> I installed the firmware update v01.08.00.00 and not having issues at the moment apart from the reprint function not working as it gets stuck on autobed leveling.
> 
> This issue also happened on the previous versions of Panda Touch.

Interesting, are you running on LAN or cloud

DubStepMad, 2025-04-29:
> > I installed the firmware update v01.08.00.00 and not having issues at the moment apart from the reprint function not working as it gets stuck on autobed leveling.
> > This issue also happened on the previous versions of Panda Touch.
> 
> Interesting, are you running on LAN or cloud

I'm running on cloud, I did have to rescan for my printer as the Touch lost access to the SD card.

dilixi, 2025-04-29:
> I installed the firmware update v01.08.00.00 and not having issues at the moment apart from the reprint function not working as it gets stuck on autobed leveling.
> 
> This issue also happened on the previous versions of Panda Touch.

Did you use ams?
Is it a multi-color printing task?
Did you use a custom filament?

What is the firmware version of your Panda Touch?

DubStepMad, 2025-04-29:
> Did you use ams? Is it a multi-color printing task? Did you use a custom filament?
>
> What is the firmware version of your Panda Touch?

- I did use AMS.
- It's a single colour task.
- I used Generic PLA filament.
- Panda Touch Version: 1.0.7.2

dilixi, 2025-04-29:
> * I did use AMS.
> * It's a single colour task.
> * I used Generic PLA filament.
> * Panda Touch Version: 1.0.7.2

Okay.
May I use your account to do a test?
Can you send me the 3mf file or a public 3mf file link which is also not working with your Panda Touch?

My email is 1284513391@qq.com.

keithwkx, 2025-04-29:
> * I did use AMS.
> * It's a single colour task.
> * I used Generic PLA filament.
> * Panda Touch Version: 1.0.7.2

Oho interesting , 🤔 wonder what will happen if custom filament and ams pro / ht is introduced. Maybe that's what caused it to only work in lan?

DubStepMad, 2025-04-29:
> > > > I installed the firmware update v01.08.00.00 and not having issues at the moment apart from the reprint function not working as it gets stuck on autobed leveling.
> > > > This issue also happened on the previous versions of Panda Touch.
> > > 
> > > 
> > > Did you use ams? Is it a multi-color printing task? Did you use a custom filament?
> > > What is the firmware version of your Panda Touch?
> > 
> > 
> > 
> > * I did use AMS.
> > * It's a single colour task.
> > * I used Generic PLA filament.
> > * Panda Touch Version: 1.0.7.2
> 
> Okay. May I use your account to do a test? Can you send me the 3mf file or a public 3mf file link which is also not working with your Panda Touch?
> 
> My email is [1284513391@qq.com](mailto:1284513391@qq.com).

Sent an email

dilixi, 2025-04-29:
> Sent an email

Sorry.
I wrote a wrong email address, I type in it by my phone.

Please send to this one:
1284513393@qq.com.

DubStepMad, 2025-04-29:
> > Sent an email
> 
> Sorry. I wrote a wrong email address, I type in it by my phone.
> 
> Please send to this one: [1284513393@qq.com](mailto:1284513393@qq.com).

Heres a link instead: [Google Drive](https://drive.google.com/drive/folders/13LTv_eiKGwVybOtwy6ETSZd_5Wogf6os?usp=sharing)

keithwkx, 2025-04-30:
@dilixi 我刚刚也更新到了固件 v1.0.8。当前配置是 P1S + AMS，使用的是 AMS 1 。暂时先不接入 HT，后续再观察是否有问题。

目前来看一切正常，AMS 自定义耗材没有受到影响(还是有识别和显示)，依然能够正常连接云端。
Panda Touch 当前使用的是 v1.7.0.2，测试了 跳过模型（Skip Object） 功能，确认可以正常使用。

问题来了，Ams Mapping好像有问题。打印文件时候选择了对应的耗材AMS没反应。

dilixi, 2025-04-30:
> [@dilixi](https://github.com/dilixi) 我刚刚也更新到了固件 v1.0.8。当前配置是 P1S + AMS，使用的是 AMS 1 。暂时先不接入 HT，后续再观察是否有问题。
> 
> 目前来看一切正常，AMS 自定义耗材没有受到影响(还是有识别和显示)，依然能够正常连接云端。 Panda Touch 当前使用的是 v1.7.0.2，测试了 跳过模型（Skip Object） 功能，确认可以正常使用。
> 
> 问题来了，Ams Mapping好像有问题。打印文件时候选择了对应的耗材AMS没反应。

谢谢反馈，
你说的打印文件的时候，是指从哪里发起的打印？

dilixi, 2025-04-30:
> I installed the firmware update v01.08.00.00 and not having issues at the moment apart from the reprint function not working as it gets stuck on autobed leveling.
> 
> This issue also happened on the previous versions of Panda Touch.

Hi,
My Panda Touch can reprint correctly.
It can finish the process of autobed leveling.

I will do the same things you did before.

Which slot did you use when you reprint it?
Did you change the filament when you try to reprint it?

dilixi, 2025-04-30:
> > > Sent an email
> > 
> > 
> > Sorry. I wrote a wrong email address, I type in it by my phone.
> > Please send to this one: [1284513393@qq.com](mailto:1284513393@qq.com).
> 
> Heres a link instead: [Google Drive](https://drive.google.com/drive/folders/13LTv_eiKGwVybOtwy6ETSZd_5Wogf6os?usp=sharing)

Hi,
My Panda Touch can reprint correctly.
It can finish the process of autobed leveling.

I will do the same things you did before.

Which slot did you use when you reprint it?
Did you change the filament when you try to reprint it?

(https://github.com/bigtreetech-beta/PandaTouch-Support/blob/main/reprint_video.mp4)

dilixi, 2025-04-30:
> [@dilixi](https://github.com/dilixi) 我刚刚也更新到了固件 v1.0.8。当前配置是 P1S + AMS，使用的是 AMS 1 。暂时先不接入 HT，后续再观察是否有问题。
> 
> 目前来看一切正常，AMS 自定义耗材没有受到影响(还是有识别和显示)，依然能够正常连接云端。 Panda Touch 当前使用的是 v1.7.0.2，测试了 跳过模型（Skip Object） 功能，确认可以正常使用。
> 
> 问题来了，Ams Mapping好像有问题。打印文件时候选择了对应的耗材AMS没反应。

是有问题，他们改了这里的逻辑，Panda Touch需要再更新一下适配它

dilixi, 2025-04-30:
> [@dilixi](https://github.com/dilixi) 我刚刚也更新到了固件 v1.0.8。当前配置是 P1S + AMS，使用的是 AMS 1 。暂时先不接入 HT，后续再观察是否有问题。
> 
> 目前来看一切正常，AMS 自定义耗材没有受到影响(还是有识别和显示)，依然能够正常连接云端。 Panda Touch 当前使用的是 v1.7.0.2，测试了 跳过模型（Skip Object） 功能，确认可以正常使用。
> 
> 问题来了，Ams Mapping好像有问题。打印文件时候选择了对应的耗材AMS没反应。

你好，
你是从哪里发起的打印任务的？从Panda Touch的SD卡面板吗？
是多色打印任务吗？选择了哪个AMS槽位？
新版本的加载和卸载耗材有问题，但是打印还是跟原来一样

DubStepMad, 2025-04-30:
> Hi,
> My Panda Touch can reprint correctly.
> It can finish the process of autobed leveling.
>
> I will do the same things you did before.
>
> Which slot did you use when you reprint it?
> Did you change the [...]

I used slot 3 of the AMS and didn't change the filament.

dilixi, 2025-04-30:
> I used slot 3 of the AMS and didn't change the filament.

https://github.com/bigtreetech-beta/PandaTouch-Support/blob/main/reprint_video.mp4)
You can check this video first.

I can reprint [this file](https://drive.google.com/drive/folders/13LTv_eiKGwVybOtwy6ETSZd_5Wogf6os?usp=sharing) successful.

What is the color when you reprint it?
Did the color is same as my video?

dilixi, 2025-04-30:
> I used slot 3 of the AMS and didn't change the filament.

Did you use [this file](https://drive.google.com/drive/folders/13LTv_eiKGwVybOtwy6ETSZd_5Wogf6os?usp=sharing) and reprint it failed?

May I use your account to do a test?

dilixi, 2025-04-30:
> Is it safe to update P1S with new Firmware without affecting the Panda Touch?
>
> version 01.08.00.00 update information :
>
> # Things to know before updating
> In order to use all the new features [...]

Hi,
Panda Touch can not control the AMS anymore if you updating the printer to 01.08.00.00.

But.
There is a new firmware you can try it after updating the printer to 01.08.00.00.

https://github.com/bigtreetech-beta/PandaTouch-Support/blob/main/firmware/v1.0.7.2_f.bin

keithwkx, 2025-04-30:
> > [@dilixi](https://github.com/dilixi) 我刚刚也更新到了固件 v1.0.8。当前配置是 P1S + AMS，使用的是 AMS 1 。暂时先不接入 HT，后续再观察是否有问题。
> > 目前来看一切正常，AMS 自定义耗材没有受到影响(还是有识别和显示)，依然能够正常连接云端。 Panda Touch 当前使用的是 v1.7.0.2，测试了 跳过模型（Skip Object） 功能，确认可以正常使用。
> > 问题来了，Ams Mapping好像有问题。打印文件时候选择了对应的耗材AMS没反应。
> 
> 谢谢反馈， 你说的打印文件的时候，是指从哪里发起的打印？

切片发到SD卡再用Panda Touch开启任务

dilixi, 2025-04-30:
> 切片发到SD卡再用Panda Touch开启任务

你是说你切片后发起了打印任务传到的SD卡吗？
打印完了再使用Panda Touch重新打印？
还是说你切片后
把SD卡从打印机上拔下来，然后切片软件再发送到SD卡，再把SD卡插回到打印机，然后再从Panda Touch打印？

你可以试一下，但是我没有改AMS mapping的逻辑，也可以正常打印，可能你那边情况又不一样。
https://github.com/bigtreetech-beta/PandaTouch-Support/blob/main/firmware/v1.0.7.2_f.bin

Disttrack, 2025-04-30:
> Hi, Panda Touch can not control the AMS anymore if you updating the printer to 01.08.00.00.
>
> But. There is a new firmware you can try it after updating the printer to 01.08.00.00. [...]

With the new PandaTouch firmware you mentioned (v1.0.7.2_f), as well as the P1S v01.08, I can correctly remap the AMS filaments, even the custom ones.

dilixi, 2025-04-30:
> With the new PandaTouch firmware you mentioned (v1.0.7.2_f), as well as the P1S v01.08, I can correctly remap the AMS filaments, even the custom ones.

Thanks for your testing and response.

DubStepMad, 2025-04-30:
> Hi, Panda Touch can not control the AMS anymore if you updating the printer to 01.08.00.00.
>
> But. There is a new firmware you can try it after updating the printer to 01.08.00.00. [...]

Using v1.0.7.2_f firmware I still run into the same issue when clicking 'reprint' from the print history on the Panda Touch.

The PS1 printer starts to process it but when it gets to the auto levelling, the toolhead is over the purge chute, heated up but with no fan on.
Also the bed doesn't move in this state, even after leaving it for around 10 minutes.

keithwkx, 2025-04-30:
> Using v1.0.7.2_f firmware I still run into the same issue when clicking 'reprint' from the print history on the Panda Touch.
>
> The PS1 printer starts to process it but when it gets to the auto [...]

What slicer are you using

DubStepMad, 2025-04-30:
> > > > > > Is it safe to update P1S with new Firmware without affecting the Panda Touch?
> > > > > > version 01.08.00.00 update information :
> > > > > > # Things to know before updating
> > > > > > In order to use all the new features introduced by the new firmware update, Bambu Studio and Bambu Handy should be updated to the latest version available:
> > > > > > 
> > > > > > * Bambu Studio: v2.0.3.54 or later
> > > > > > * Bambu Handy: v3.1.0 or later
> > > > > > 
> > > > > > # New Features
> > > > > > 
> > > > > > 1. Added support for multi-plate printing
> > > > > > 
> > > > > > 
> > > > > > * You can now select "Print all" from the Bambu Studio interface, or you can use the Multi-Plate printing operation from the printer's screen.
> > > > > > 
> > > > > > 
> > > > > > 2. Added support for AMS 2 Pro and AMS HT
> > > > > > 
> > > > > > 
> > > > > > * The printer now supports starting the filament drying operation from the printer's screen
> > > > > > 
> > > > > > 
> > > > > > 3. Added Portuguese language
> > > > > > 4. Added Korean language
> > > > > > 
> > > > > > # Note
> > > > > > 
> > > > > > 1. Please update the printer firmware before connecting the printer to the AMS 2
> > > > > >    Pro/HT
> > > > > > 2. Each P1 series printer supports up to four AMS units in total, regardless of type.
> > > > > >    You can mix and match different AMS models, such as the AMS, AMS 2 Pro, and AMS HT, within this limit.Firm
> > > > > 
> > > > > 
> > > > > Hi, Panda Touch can not control the AMS anymore if you updating the printer to 01.08.00.00.
> > > > > But. There is a new firmware you can try it after updating the printer to 01.08.00.00.
> > > > > https://github.com/bigtreetech-beta/PandaTouch-Support/blob/main/firmware/v1.0.7.2_f.bin
> > > > 
> > > > 
> > > > Using v1.0.7.2_f firmware I still run into the same issue when clicking 'reprint' from the print history on the Panda Touch.
> > > > The PS1 printer starts to process it but when it gets to the auto levelling, the toolhead is over the purge chute, heated up but with no fan on. Also the bed doesn't move in this state, even after leaving it for around 10 minutes.
> > > 
> > > 
> > > What slicer are you using
> > 
> > 
> > Bambu Lab Studio - v2.0.3.54
> 
> So, you don't wan to send me your account to do a test?
> 
> I did a test with Generic PLA and It can work well.

What account are you referring to?

I'm not sharing login information if that's what you mean.

dilixi, 2025-04-30:
> > > > > Is it safe to update P1S with new Firmware without affecting the Panda Touch?
> > > > > version 01.08.00.00 update information :
> > > > > # Things to know before updating
> > > > > In order to use all the new features introduced by the new firmware update, Bambu Studio and Bambu Handy should be updated to the latest version available:
> > > > > 
> > > > > * Bambu Studio: v2.0.3.54 or later
> > > > > * Bambu Handy: v3.1.0 or later
> > > > > 
> > > > > # New Features
> > > > > 
> > > > > 1. Added support for multi-plate printing
> > > > > 
> > > > > 
> > > > > * You can now select "Print all" from the Bambu Studio interface, or you can use the Multi-Plate printing operation from the printer's screen.
> > > > > 
> > > > > 
> > > > > 2. Added support for AMS 2 Pro and AMS HT
> > > > > 
> > > > > 
> > > > > * The printer now supports starting the filament drying operation from the printer's screen
> > > > > 
> > > > > 
> > > > > 3. Added Portuguese language
> > > > > 4. Added Korean language
> > > > > 
> > > > > # Note
> > > > > 
> > > > > 1. Please update the printer firmware before connecting the printer to the AMS 2
> > > > >    Pro/HT
> > > > > 2. Each P1 series printer supports up to four AMS units in total, regardless of type.
> > > > >    You can mix and match different AMS models, such as the AMS, AMS 2 Pro, and AMS HT, within this limit.Firm
> > > > 
> > > > 
> > > > Hi, Panda Touch can not control the AMS anymore if you updating the printer to 01.08.00.00.
> > > > But. There is a new firmware you can try it after updating the printer to 01.08.00.00.
> > > > https://github.com/bigtreetech-beta/PandaTouch-Support/blob/main/firmware/v1.0.7.2_f.bin
> > > 
> > > 
> > > Using v1.0.7.2_f firmware I still run into the same issue when clicking 'reprint' from the print history on the Panda Touch.
> > > The PS1 printer starts to process it but when it gets to the auto levelling, the toolhead is over the purge chute, heated up but with no fan on. Also the bed doesn't move in this state, even after leaving it for around 10 minutes.
> > 
> > 
> > can you send me a shotscreen of the filaments on panda touch?
> 
> [image]
> 
> Currently printing the task it couldn't reprint via Panda Touch.

can you change the filament type to PETG or something else on Panda Touch and have a try again?

dilixi, 2025-04-30:
> [image]
>
> Currently printing the task it couldn't reprint via Panda Touch.

I think you can change your password to a simple one and you can change back when I solve this problem for you.
That is a very fast way.

Disttrack, 2025-05-01:
With this version of OrcaSlicer in a new PR, https://github.com/SoftFever/OrcaSlicer/pull/9517 I can use AMS mapping in Pandatouch and reprint from history and normal reprint from "printing screen".

dilixi, 2025-05-01:
> With this version of OrcaSlicer in a new PR, [SoftFever/OrcaSlicer#9517](https://github.com/SoftFever/OrcaSlicer/pull/9517) I can use AMS mapping in Pandatouch and reprint from history and normal reprint from "printing screen".

Thanks.

I have reprinted the file which he send to me. And everything was fine.

Maybe there is something different.

Noisyfox, 2025-05-01:
To make AMS mapping work on new firmware, the 3mf file must be updated with the new `filament_maps` property, and the network command with the new `ams_mapping2` proterty. Otherwise the printer will stuck at the "preheating bed" step and won't continue.

keithwkx, 2025-05-01:
> To make AMS mapping work on new firmware, the 3mf file must be updated with the new `filament_maps` property, and the network command with the new `ams_mapping2` property. Otherwise the printer will stuck at the "preheating bed" step and won't continue.

Thank you for the clarification and for your dedicated work on OrcaSlicer

skl111, 2025-05-03:
I am not sure if this is a known bug or issue introduced with this firmware, but when using the Orca Dev 2.3.1 on a P1 series printer with only the eternal spool, and you set your Filament to say PETG and then set a calibration K value from the list to say .036, it will revert back to the default .020 within 60 seconds, it happens in Orca 2.3.1 and BS 1.10.2.76. reverting back to 1.0.7 firmware fixes this.
I am thinking it maybe due to the new network plugin or a new command implemented to update the K value.

BradCOnGitHub, 2025-05-07:
> It seems as if this version is safe to use based on reports we have been getting but we are still waiting for it to become available on our machines to verify.

Any official update on this yet? I don't think there's any issues but there's nothing in this firmware that helps me so I'm holding off until I hear it's been properly tested.

Noisyfox, 2025-05-07:
> I am thinking it maybe due to the new network plugin or a new command implemented to update the K value.

yes, please try https://github.com/SoftFever/OrcaSlicer/pull/9517

skl111, 2025-05-08:
> > I am thinking it maybe due to the new network plugin or a new command implemented to update the K value.
> 
> yes, please try [SoftFever/OrcaSlicer#9517](https://github.com/SoftFever/OrcaSlicer/pull/9517)

Ok, so that keeps the value using the new firmware, with the old firmware i have KR PETG set to .36 on the printer but when I go to edit it, it shows a different filament brand and the default value, even though it sees the K value set correctly.

[image]

Noisyfox, 2025-05-08:
> > > I am thinking it maybe due to the new network plugin or a new command implemented to update the K value.
> > 
> > 
> > yes, please try [SoftFever/OrcaSlicer#9517](https://github.com/SoftFever/OrcaSlicer/pull/9517)
> 
> Ok, so that keeps the value using the new firmware, with the old firmware i have KR PETG set to .036 on the printer but when I go to edit it, it shows a different filament brand and the default value, and an option to select that filaments K value, even though it sees the K value set correctly with the KR PETG filament.
> 
> [image]

Old version can only set the very first ams slot correctly. Or should I say, it set the first slot regardless of which one you actually chose from the slicer.

Nobiscum, 2025-05-20:
I dont find there a solution so this is my case: I use external spool and there is no "reprint" button. I can start printing only from the print history.

andyspoo, 2025-06-28:
> `filament_maps` property,

Are you talking about manually editing the .3mf file or can you get Bambu Studio to add it  automatically?