bigtreetech/PandaTouch #300: 404 when downloading Firmware

scobby, 2025-05-07:
@bigtreetech @dilixi  
the Firmware Link in the Release is a 404 Error. Please fix

scobby, 2025-05-07:
[image]

dilixi, 2025-05-07:
> <img alt="Image" width="1797" src="https://private-user-images.githubusercontent.com/9264361/441087929-e7cf6bad-b3c0-441e-8c99-a46600d27845.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDY1OTkzMjUsIm5iZiI6MTc0NjU5OTAyNSwicGF0aCI6Ii85MjY0MzYxLzQ0MTA4NzkyOS1lN2NmNmJhZC1iM2MwLTQ0MWUtOGM5OS1hNDY2MDBkMjc4NDUucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDUwNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTA1MDdUMDYyMzQ1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZTdlYzM3ZWQ3ZWIxYWY2ODc1MWFmM2ZhNjJhZThhYzQxOWQzODY4OTBkYTk4MDBkYmQ2YjFmMmYwZTg0ZDNhZCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.76d_lLe-39bRRg3pYWQtDSEWKrzAYNPHtk-xG85opPc">

I can download it successful.

1.0.7.1 has a bug.

You can try this one.

[1.0.7.2.zip](https://github.com/user-attachments/files/20077403/1.0.7.2.zip)