bigtreetech/PandaTouch #301: P1S new firmware version 01.08.00.01 Beta released - Careful - Panda touch won't work with it.

Winky1st, 2025-05-09:
Hi 
As per title, a new firmware version 01.08.00.01 Beta has been released.
Panda touch won't work with it..

In case you can revert back to fw 01.08.00.00 from bambu handy
Cheers

bigtreetech, 2025-05-09:
We just saw this too and are preparing some social media announcements. Thanks for creating the issue.

bigtreetech, 2025-05-09:
This is a complex question to answer. Fundamentally, Bambu are trying to do something that known to be impossible in the security world. They are trying to distribute secrets in software onto non-secure hardware and keep them secure through obscurity. It is my personal belief that those secrets will continue to be extracted which will deem the entire authentication mechanism null and void since anyone will be able to request certificates and the respective private keys. Whether or not we choose to use that secret is another story. We are still debating this.

Winky1st, 2025-05-09:
Ty for your perfect explanation.
I hope they will come to their senses and turn back from the wrong steps they have taken.

bl4ckm0onk3y, 2025-05-10:
Well, that's it for me, time to set the LAN Mode and be dont with it, not worth imho, next printers i'll buy will be klipper based.

JakobTewes, 2025-05-11:
Also sticking in here. Pretty interested, on how you decide whether to "go for secret extraction" ;-)

Winky1st, 2025-05-12:
They both well Prepared and Honorable Companies, Ima pretty sure they will find a speech, a dialogue. 
At the end BTT is doing lot of nice aftermarket tools&devices for their printers so it seems to be interesting market for both them. 
But this is only imho!.

dadzikowski, 2025-05-15:
Please find a way to continue to have the Panda Touch work. Love your products and I use my Panda Touch all the time. I really don't want to go LAN-only because I will loose camera access, etc.

Enclave10, 2025-05-16:
> Пожалуйста, найдите способ, чтобы Panda Touch продолжал работать. Мне нравятся ваши продукты, и я постоянно пользуюсь Panda Touch. Я действительно не хочу переходить на работу только по локальной сети, потому что я потеряю доступ к камере и т.
 камера и в локальном работает

keithwkx, 2025-05-20:
weird just updated to 1.08.01.00 panda touch works as normal

andyspoo, 2025-06-28:
> This is a complex question to answer. Fundamentally, Bambu are trying to do something that is known to be impossible in the security world. They are trying to distribute secrets in software onto non-secure hardware and keep them secure through obscurity. It is my personal belief that those secrets will continue to be extracted which will deem the entire authentication mechanism null and void since anyone will be able to request certificates and the respective private keys. Whether or not we choose to use that secret is another story. We are still debating this.

Bamdu printers were easily the number one choice and then they knocked their own brand by trying to be clever with security.  It's ridiculous, especially when people will refuse to upgrade their firmware to loose features they want. Firmware that might include security fixes !!! They are actually making their printers less secure in the long run.