bigtreetech/PandaTouch #303: AMS PRO 2 dying options

Melficeapollo, 2025-05-12:
with the new update, are we going to get the function of drying filament through the AMS rather than using the print bed?

also, when will we be getting the feature to change the BTT logo on the main screen?

Neuropsihopuppu, 2025-05-16:
following because im also intrested in the answear for this 2 questions

ssikssik, 2025-05-18:
https://ptimgtool.bttwiki.com/
You can change the logo here

Neuropsihopuppu, 2025-05-18:
> https://ptimgtool.bttwiki.com/ You can change the logo here

Not that logo.

We are talking about ths big logo on the home/idle screen

bigtreetech, 2025-06-14:
It'll be coming.

Rey116bro, 2025-07-11:
> It'll be coming.

We are really looking forward to that since that is holding me back to a bigtreetech panda touch

Melficeapollo, 2025-08-12:
Doesn't seem like it, they are too busy releasing ceramic funnels for the ams, metal scrapers, and more led bars.

bigtreetech, 2025-08-13:
> Doesn't seem like it, they are too busy releasing ceramic funnels for the ams, metal scrapers, and more led bars.

@Melficeapollo Terribly sorry for releasing products that have benefit and people want.

@pleszczynski We are working on a major update to the Panda Knomi and then the touch is next.

kadabriel, 2025-08-13:
Following! I need that drying option. Just set up my PandaT today, great work. And please work fast, but not so fast that it takes away from the quality of the work. You have to find that sweet spot in between, and try to be done by tomorrow morning. preferably.

bigtreetech, 2025-09-17:
Working on it.